kerberos-io/agent #15: Creating designs in Figma

cedricve, 2020-12-29:
Started creating some designs in Figma, for the different pages. Any help is welcome!
https://www.figma.com/file/PpjFz2GGX6ZaWyIzef6mqT/Web?node-id=0%3A1

cedricve, 2020-12-30:
Currently working on the login page: figma.com/file/PpjFz2GGX6ZaWyIzef6mqT/Web?node-id=57%3A0

cedricve, 2021-04-11:
Designs are done, thanks @AllienWorks. All designs can be viewed here: https://www.figma.com/file/msuYC6sv2cOCqZeDtBxNy7/NEW-Kerberos.io-Apps?node-id=1%3A2

AllienWorks, 2021-04-14:
Hello everybody,

if you're interested to play with the latest prototype, here's the latest version:

## [Latest Kerberos.io Open Source Agent prototype](https://www.figma.com/proto/msuYC6sv2cOCqZeDtBxNy7/Kerberos.io-Apps?node-id=1%3A1464&viewport=768%2C361%2C0.16784118115901947&scaling=min-zoom&page-id=1%3A2%3Ffuid%3D449684443467913607)

It's a clickable prototype, mimicking the real behaviour. Of course it's quite limited, but it will probably give you a better feeling of the Open Source Agent anyway :)

closed 2022-09-08 by cedricve