kerberos-io/agent #27: What is happening on port 80?

Obezyana33, 2021-09-14:
I am running 7 Podman containers, which all stream to Kerberos Cloud.  

When locking down my CCTV VLAN and firewall, I noticed that the kerberos.io containers attempt connections to the following hosts:

| IP Address         | Protocol | Port |
|:----------------------|:------------:|--------|
| 159.65.215.225 | TCP         | 1883 |
|172.67.133.190 | TCP          | 443 |
|104.155.97.237 | TCP          | 443 |
|34.79.161.171  | TCP           |  80 |

Now TCP Port 443 is not in question - I assume that kerberos.io containers push their streams via 443 to the above two servers.  1883 is MQTT and Port 80 is unencrypted HTTP!

My questions are as follows:
1. Why are the containers connecting to 34.79.161.171 over port 80?  What is this connection being used for?
2. What is MQTT port 1883 being used for specifically?

FYI, I have been unable to find these answers in the documentation, faq, or list of resolved issues.

cedricve, 2021-09-14:
You can verify the source code here: https://github.com/kerberos-io/machinery.

This is a duplicate of https://github.com/kerberos-io/machinery/issues/199.

Obezyana33, 2021-09-14:
Thanks for the quick response Cédric, I appreciate your work and love the product!

Unfortunately I am not a coder, so searching through source code to figure out what servers do on what ports, is not really in my wheelhouse. I wouldn't know where to start!

The duplicate issue answers my question regarding MQTT (thank you), but not the more critical question regarding 34.79.161.171:80.  If containers are sending unencrypted data from my sensitive CCTV streams over the internet, I would like to know what that data is and where it is being sent.

As a paying customer, can I raise a support ticket?  Or is there some other way that I can find my answer?

cedricve, 2021-09-14:
no worries @Obezyana33 we should address these questions in public, so other users also know this.

For 34.79.161.171:80 this is what going on:  https://github.com/kerberos-io/machinery/blob/master/src/kerberos/cloud/Cloud.cpp#L179-L193

It is some polling mechanism that we've build in the past for Raspberry Pi's on a WIFI connection. A couple of years ago then we had WIFI dongles that were timing out when no activity, and therefore no longer forwarded recordings to our cloud application. Over time we should have get rid of that code polling, as it is no more relevant. 

The only data that is send to this endpoint is the following: https://github.com/kerberos-io/machinery/blob/master/src/kerberos/cloud/Cloud.cpp#L188-L193. We tried to use this data in the past to gather analytics of how many agents are on which version, however we no longer do any analytics, and this endpoint is just a ping/pong api. Disabling this IP address will not affect any feature or function.

We will never send sensitive data to any cloud solution, without documenting the properly.

Obezyana33, 2021-09-15:
Thanks Cédric.  Your answer has put my mind at ease!

closed 2021-09-15 by Obezyana33