kerberos-io/agent #32: Feature request: Raw recording compression

GHNewbiee, 2022-05-27:
Nice to have such a feature to reduce the size of the storage means.

cedricve, 2022-09-08:
@GHNewbiee  how would you see this? Any feedback?

olokos, 2022-09-09:
Maybe we could have a config value named "archive old recordings every X hours/days"?

After those x hours, there could be a zip/7zip archive created with all of the recordings made after X hours.

So all of the old recordings could be compressed in their individual archive for a given day, considering they're older than X days, just an idea that could work well.

I already do this myself, after each day I create a new folder in /recordings and move all of the old recordings there.

With this done programmatically, instead of moving the recordings to folders, we could instead compress them into archives instead.

I suggest not using Linux timestamps for archives but regular date timestamps, or at least having an option to choose datetime format.
Linux timestamps are epic for individual files, but would make those archives harder to figure out which day was the archive made on.

cedricve, 2022-09-09:
wonderful idea! im just wondering what is the compression ratio you are currently experiencing? Like the total size before and after zipping..

olokos, 2022-09-09:
I just tried on windows 7zip ultra and zip ultra various settings and I was hardly getting size reduction, I guess h264 is compressed enough already.

Maybe with `tar` on linux results will be better?
If not then maybe at least creating/choosing a directory and then moving old recordings there periodically?

Some archiving feature would be amazing, if we can't get it compressed any better with reasonable speed.

cedricve, 2022-09-10:
Thanks I would give tar a try, however the whole idea is to use cloud storage here for backing up, and freeing up disk space. 

This cloud can also be a synology at home. So that would mean you are moving recordings from the agent to your local redundant storage.

Kind regards,
Verstraeten Cédric

bigntallmike, 2022-12-16:
There's no reason to believe that archiving compressed video would save space in this way. The best way to save space with compressed video is to use a better codec or a lower bit-rate. The redundancies standard compression algorithms would find should already be gone.
Re-encoding with H265 would make a huge difference to storage though.