kerberos-io/agent #34: Live Stream/View Framerate

tokenwizard, 2022-06-10:
I'm running the latest Kerberos.io Agent via Docker on an Ubuntu 20 Server VM minimal install. That VM has 4 cores and 8GB of RAM dedicated. I was able to connect my S3VC PoE IP Camera via RTSP using the primary RTSP stream. The Steam is set to 3MP resolution with 20fps H265. 

I have the Streaming Framerate set in the Agent to 20fps to match the RTSP stream. 

The clips captured via Motion Detection in the Agent playback at the expected framerate, but the live view on the dashboard and the Stream provided via the Agent run at about 1/2fps, not even 1fps. I have tried dumbing down the RTSP stream to 1280x720 H264 and it still only runs about 1/2fps on the Live Stream. My plan was to plug several of the Live Streams from multiple cameras into a simple HTML Table to be able to view them all live simultaneously, but at this rate the Live Stream is running, this is not going to be viable. 

Is there something I'm missing? 

What logs can I pull to assist in troubleshooting?

Thanks!

olokos, 2022-09-08:
As far as I know, agent currently supports only h264 hardware acceleration.

H265 encoding done by software is very very slow, which might be the cause of your issues, especially if You're running this in a vm, that is running docker and has no GPU pass through to VM for H265 HW encoding.

Even if you manage to get h265 GPU HW accelerated, I'm not sure if RTSP stream supports H265 hardware encoding and then if agent itself can do h265 hardware decoding.

IIRC only H264 hardware encoding currently works, as there's absolutely no Raspberry Pi in the world that supports H265 HW, but Pi4B does support H264 HW.

cedricve, 2022-09-08:
This is related to the old agent (open source). Within the agent we now support WEBRTC which gives a better performance in combination with [Kerberos Hub SAAS](https://app.kerberos.io).

closed 2022-09-08 by cedricve