kerberos-io/agent #40: panic: invalid memory pointer on nil pointer dereference after #39

olokos, 2022-08-18:
[image]
```
{"level":"info","msg":"FindMotion: Number of changes detected:9455","time":"2022-08-18T12:17:22+02:00"}
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x227a00]

goroutine 66 [running]:
github.com/kerberos-io/joy4/format/mp4.(*Muxer).WriteHeader(0x0, {0x24101b0, 0x1, 0x1})
        /go/pkg/mod/github.com/kerberos-io/joy4@v1.0.33/format/mp4/muxer.go:154 +0x18
github.com/kerberos-io/agent/machinery/src/capture.HandleRecordStream(0x24a6910, 0x273ec00, 0x2700360, {0x24101b0, 0x1, 0x1})
        /go/src/github.com/kerberos-io/agent/machinery/src/capture/main.go:256 +0x6f4
created by github.com/kerberos-io/agent/machinery/src/components.RunAgent
        /go/src/github.com/kerberos-io/agent/machinery/src/components/Kerberos.go:109 +0x974

```

Last night it was pitch black when I was working on this, so there was nothing to detect, but now I came to dead docker container, with the reason behind it listed above.

The screenshot and the text logs are exactly the same, just provided them in both forms, so ez copy paste and easy to read at the same time, thanks to colors on ss.

I will investigate this, but it's related to joy4, which I noticed that @thomasqp did write a comment on top of some of the files that it needs to be updated/changed, I'm not sure myself, but I will look into this.

cedricve, 2022-08-18:
I believe this is introduced by not being able to create the muxer. https://github.com/kerberos-io/agent/blob/develop/machinery/src/capture/main.go#L245-L256

Some additional logging and error checking might explain what is going on.

olokos, 2022-08-18:
This bug was caused by no write permissions from the docker container into the local filesystem.

To be more specific, in this PR Dockerfile changed from `#USER agent` into `USER agent`.

the `user` which I use to create the docker image is in `docker` group
I also have a local account `agent` which is in the group `user`

locally mounted `recordings` directory had `775` permissions, which caused that panic bug.
with `chmod 777 recordings` this bug went away.

Perhaps better error handling here would be great, as right now I wasn't able to realise that this entire issue was just no write permissions, basically.

cedricve, 2022-08-18:
good point we'll need to clarrify this in the documentation, and provide better error handling in the codebase

closed 2022-08-18 by cedricve

thomasqp, 2022-10-11:
Can you simulate this error?

Kind regards,
Verstraeten Cédric