kerberos-io/agent #42: Something went wrong while verifying storage credentials: {"code":401,"message":"cookie token is empty"}

nurio-felix, 2022-09-07:
[image]

Please, it is possible check this?

cedricve, 2022-09-07:
amazing you are already using it..
Can you verify if the endpoint is valid? You dont need to add the suffix "storage" anymore.

Kind regards,
Verstraeten Cédric

cedricve, 2022-09-08:
@nurio-felix do you need more help on this?

nurio-felix, 2022-09-08:
Hi, thx for your reply. 
Is working fine now.
The error was in the vault, Accounts should have "*" in directory

[image]

Now I am having another error in "Continuous recording".
I will open new issue for this.

closed 2022-09-11 by cedricve