kerberos-io/agent #44: panic: runtime error: invalid memory address or nil pointer dereference

nurio-felix, 2022-09-08:
Hello,

I am trying enable "Continuous recording" but no success.
Every time I enable contianer crash.

[GIN-debug] [WARNING] Creating an Engine instance with the Logger and Recovery middleware already attached.

[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:   export GIN_MODE=release
 - using code:  gin.SetMode(gin.ReleaseMode)

[GIN-debug] GET    /debug/pprof/             --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/cmdline      --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/profile      --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] POST   /debug/pprof/symbol       --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/symbol       --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/trace        --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/allocs       --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/block        --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/goroutine    --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/heap         --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/mutex        --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/threadcreate --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
{"level":"info","msg":"RunAgent: opening RTSP stream","time":"2022-09-08T16:10:58+01:00"}
[GIN-debug] GET    /swagger/*any             --> github.com/swaggo/gin-swagger.CustomWrapHandler.func1 (4 handlers)
[GIN-debug] GET    /config                   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func1 (4 handlers)
[GIN-debug] POST   /config                   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func2 (4 handlers)
[GIN-debug] POST   /api/login                --> github.com/appleboy/gin-jwt/v2.(*GinJWTMiddleware).LoginHandler-fm (4 handlers)
[GIN-debug] GET    /api/dashboard            --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func3 (4 handlers)
[GIN-debug] POST   /api/latest-events        --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func4 (4 handlers)
[GIN-debug] GET    /api/days                 --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func5 (4 handlers)
[GIN-debug] GET    /api/stream               --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func6 (4 handlers)
[GIN-debug] GET    /api/config               --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func7 (4 handlers)
[GIN-debug] POST   /api/config               --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func8 (4 handlers)
[GIN-debug] GET    /api/restart              --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func9 (4 handlers)
[GIN-debug] GET    /api/stop                 --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func10 (4 handlers)
[GIN-debug] POST   /api/hub/verify           --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func11 (4 handlers)
[GIN-debug] POST   /api/persistence/verify   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func12 (4 handlers)
[GIN-debug] GET    /file/*filepath           --> github.com/gin-gonic/gin.(*RouterGroup).createStaticHandler.func1 (9 handlers)
[GIN-debug] HEAD   /file/*filepath           --> github.com/gin-gonic/gin.(*RouterGroup).createStaticHandler.func1 (9 handlers)
[GIN-debug] [WARNING] You trusted all proxies, this is NOT safe. We recommend you to set a value.
Please check https://pkg.go.dev/github.com/gin-gonic/gin#readme-don-t-trust-all-proxies for details.
[GIN-debug] Listening and serving HTTP on :8080
{"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2022-09-08T16:10:58+01:00"}
{"level":"info","msg":"ConfigureMQTT: Set broker uri ","time":"2022-09-08T16:10:59+01:00"}
{"level":"info","msg":"ConfigureMQTT: Set username xxx","time":"2022-09-08T16:10:59+01:00"}
{"level":"info","msg":"ConfigureMQTT: Set password xxx","time":"2022-09-08T16:10:59+01:00"}
{"level":"error","msg":"HandleHeartBeat: (400) Something went wrong while sending to Kerberos Hub.","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"HandleLiveStreamHD: Waiting for peer connections.","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"HandleRecordStream: Start continuous recording ","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"WriteToTrack: listing codecs.","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"WriteToTrack: codec - H264 found.","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"H264","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"WriteToTrack: codec - PCM_ALAW found.","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"PCM_ALAW","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"WriteToTrack: not using a transcoder.","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"ProcessMotion: Continuous recording, so no motion detection.","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"HandleStream: packet size 176627","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"Recording started","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"HandleRecordStream: composing recording","time":"2022-09-08T16:11:02+01:00"}
{"level":"info","msg":"HandleRecordStream: write header","time":"2022-09-08T16:11:02+01:00"}
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x28 pc=0x6196a0]

goroutine 30 [running]:
github.com/kerberos-io/joy4/format/mp4.(*Muxer).WriteHeader(0xc000034050?, {0xc00046eaa0?, 0x242?, 0xc00041eec0?})
        /go/pkg/mod/github.com/kerberos-io/joy4@v1.0.33/format/mp4/muxer.go:154 +0x20
github.com/kerberos-io/agent/machinery/src/capture.HandleRecordStream(0xc00003c620?, 0xc0001e6000, 0xc0005d6070, {0xc00046eaa0, 0x2, 0x2})
        /go/src/github.com/kerberos-io/agent/machinery/src/capture/main.go:194 +0x18fb
created by github.com/kerberos-io/agent/machinery/src/components.RunAgent
        /go/src/github.com/kerberos-io/agent/machinery/src/components/Kerberos.go:125 +0x8ef`
`

nurio-felix, 2022-09-08:
**config.json**

```
{
        "type": "template",
        "key": "0x123-outdoor-camera1-agent",
        "name": "camera-outdoor1",
        "mqtt_username": "",
        "mqtt_password": "",
        "timezone": "Europe/Lisbon",
        "stunuri": "",
        "turnuri": "",
        "turn_password": "",
        "capture": {
                "id": "ipcamera",
                "ipcamera": {
                        "rtsp": "rtsp://user:pass@192.168.1.105:554/stream1",
                        "fps": "15"
                },
                "usbcamera": {
                        "device": "",
                        "fps": ""
                },
                "raspicamera": {
                        "device": "",
                        "fps": ""
                },
                "continuous": "true",
                "postrecording": 10,
                "prerecording": 5,
                "maxlengthrecording":60,
                "forwardwebrtc": "",
                "transcodingwebrtc": "false",
                "transcodingresolution": 25,
                "fragmented": "false",
                "fragmentedduration": 1
        },
        "timetable": [
                {
                        "start1": 0,
                        "end1": 43199,
                        "start2": 43200,
                        "end2": 86400
                },
                {
                        "start1": 0,
                        "end1": 69840,
                        "start2": 70080,
                        "end2": 86400
                },
                {
                        "start1": 0,
                        "end1": 43199,
                        "start2": 43200,
                        "end2": 86400
                },
                {
                        "start1": 0,
                        "end1": 43199,
                        "start2": 43200,
                        "end2": 86400
                },
                {
                        "start1": 0,
                        "end1": 43199,
                        "start2": 43200,
                        "end2": 86400
                },
                {
                        "start1": 0,
                        "end1": 43199,
                        "start2": 43200,
                        "end2": 86400
                },
                {
                        "start1": 0,
                        "end1": 43199,
                        "start2": 43200,
                        "end2": 86400
                }
        ],
        "region": {
                "rectangle": {
                        "x1": 0,
                        "y1": 0,
                        "x2": 800,
                        "y2": 640
                },
                "polygon": []
        },
        "cloud": "kstorage",
        "hub_uri": "",
        "hub_key": "",
        "hub_private_key": "",
        "hub_site": "",
        "kstorage": {
                "uri": "https://kerberos-vault-api.domain.tld",
                "access_key": "access_key",
                "secret_access_key": "secret_access_key",
                "provider": "storj",
                "directory": ""
        }
}

```

Every time I need update something need write directly in config.json because from web is not working/saving to config.json file

cedricve, 2022-09-08:
>                         "rtsp": ***@***.***:554/stream1",

you have an old config can you start over, we just released a new version. pull the container with latest version.

Kind regards,
Verstraeten Cédric

nurio-felix, 2022-09-08:
I pulled **kerberos/agent-dev:latest**, but error remains same:

Also updating in web app still not write to config.json

```
[GIN-debug] [WARNING] You trusted all proxies, this is NOT safe. We recommend you to set a value.
Please check https://pkg.go.dev/github.com/gin-gonic/gin#readme-don-t-trust-all-proxies for details.
[GIN-debug] Listening and serving HTTP on :8080
{"level":"info","msg":"ConfigureMQTT: Set broker uri tcp://mqtt.kerberos.io:1883","time":"2022-09-08T20:08:24+01:00"}
{"level":"info","msg":"ProcessMotion: Continuous recording, so no motion detection.","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"HandleRecordStream: Start continuous recording ","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"HandleLiveStreamHD: Waiting for peer connections.","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"WriteToTrack: listing codecs.","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"WriteToTrack: codec - H264 found.","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"H264","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"WriteToTrack: codec - PCM_ALAW found.","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"PCM_ALAW","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"WriteToTrack: not using a transcoder.","time":"2022-09-08T20:08:25+01:00"}
{"level":"error","msg":"HandleHeartBeat: (400) Something went wrong while sending to Kerberos Hub.","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"HandleStream: packet size 257723","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"Recording started","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"HandleRecordStream: composing recording","time":"2022-09-08T20:08:25+01:00"}
{"level":"info","msg":"HandleRecordStream: write header","time":"2022-09-08T20:08:25+01:00"}
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x28 pc=0x6196a0]

goroutine 52 [running]:
github.com/kerberos-io/joy4/format/mp4.(*Muxer).WriteHeader(0xc0010cc2d0?, {0xc00044ed00?, 0x242?, 0xc001064c00?})
        /go/pkg/mod/github.com/kerberos-io/joy4@v1.0.33/format/mp4/muxer.go:154 +0x20
github.com/kerberos-io/agent/machinery/src/capture.HandleRecordStream(0xc00003c5c0?, 0xc0001ba000, 0xc0004be310, {0xc00044ed00, 0x2, 0x2})
        /go/src/github.com/kerberos-io/agent/machinery/src/capture/main.go:194 +0x18fb
created by github.com/kerberos-io/agent/machinery/src/components.RunAgent
        /go/src/github.com/kerberos-io/agent/machinery/src/components/Kerberos.go:125 +0x8ef

```

**New config.json**

```
{
        "type": "",
        "key": "outdoor-camera1-agent",
        "name": "agent",
        "time": "false",
        "offline": "false",
        "auto_clean": "true",
        "max_directory_size": 100,
        "timezone": "Europe/Lisbon",
        "capture": {
                "name": "",
                "ipcamera": {
                        "rtsp": "rtsp://user:pass@192.168.1.92:105/stream1",
                        "sub_rtsp": "",
                        "fps": "15"
                },
                "usbcamera": {
                        "device": ""
                },
                "raspicamera": {
                        "device": ""
                },
                "continuous": "true",
                "postrecording": 20,
                "prerecording": 10,
                "maxlengthrecording": 30,
                "transcodingwebrtc": "",
                "transcodingresolution": 0,
                "forwardwebrtc": "",
                "fragmented": "false",
                "fragmentedduration": 8,
[... 55 lines cut ...]
        },
        "cloud": "kstorage",
        "s3": {},
        "kstorage": {
                "uri": "https://kerberos-vault-api.domain.tld",
                "access_key": "access_key",
                "secret_access_key": "secret_access_key",
                "provider": "storj",
                "directory": ""
        },
        "mqtturi": "tcp://mqtt.kerberos.io:1883",
        "mqtt_username": "",
        "mqtt_password": "",
        "stunuri": "stun:turn.kerberos.io:8443",
        "turnuri": "turn:turn.kerberos.io:8443",
        "turn_username": "username1",
        "turn_password": "password1",
        "heartbeaturi": "",
        "hub_uri": "https://api.cloud.kerberos.io",
        "hub_key": "",
        "hub_private_key": "",
        "hub_site": "",
        "condition_uri": ""
}
```

cedricve, 2022-09-08:
>                         "rtsp": ***@***.***:105/stream1",

Something is wrong on your side then, it should generate an automated key in your config file.
Do you have permissions correctly on the data folder? The write header might fail because it was not able to create a recording on your file system, this would explain the “save” issue you experience.

olokos, 2022-09-09:
You must set permissions of config.json to 777, also /recordings directory must be set to 777.

This error happens when there's insufficient permissions to write to directory, but there's an attempt to to write to the directory, despite no permissions.

cedricve, 2022-09-09:
We will add some validation steps at startup of the Kerberos Agent. This would make it more obvious whatbis going weong.

Kind regards,
Verstraeten Cédric

cedricve, 2022-09-11:
did that help? @nurio-felix

nurio-felix, 2022-09-12:
> did that help? @nurio-felix

@cedricve 
Sorry for delay.
Yes, I created persistent volume and now is working fine.
Thanks for help guys.

cedricve, 2022-09-12:
wonderful! we have been looking into your approach, and actually like how you like to avoid kubernetes. Would you consider to write an medium/blog article to discuss your deployment? Might be really interesting for users, to quickly bootup some agents + a Kerberos Vault.

nurio-felix, 2022-09-13:
> wonderful! we have been looking into your approach, and actually like how you like to avoid kubernetes. Would you consider to write an medium/blog article to discuss your deployment? Might be really interesting for users, to quickly bootup some agents + a Kerberos Vault.

Yes I can write docker + docker compose implementation.
I avoid kubernetes because actually I am using this for small scale in my home surveillance, so I think is more easy to setup.

For standart I think in their homes people will choose docker compose. 
It is possible making this home surveillance using Port forwarding (80,443) in Router to Virtual Machine that can be created using Virtual Box.

cedricve, 2022-11-28:
hey @nurio-felix, we are trying to consolidate this thread (and your amazing work) in our documentation, and write a dedicated blog for it. Do you have any instructions for the traefik part, is this running in a seperate container?

cedricve, 2022-11-28:
I'm consolidating it over here: https://github.com/kerberos-io/vault/tree/master/docker, but still needs some polishing. The containers are running, but the exposure (and probably signage) is missing some details.

closed 2023-02-13 by cedricve