kerberos-io/agent #47: Translation web UI (additional languages)

cedricve, 2022-09-11:
We are currently translating the Kerberos Agent UI. Find a list of the translations we are currently focusing on, feel free to add more languages if you are willing to contribute.

Languages can be found in the [public folder](https://github.com/kerberos-io/agent/tree/develop/ui/public/locales). If your language is missing, create a new directory with the language 2 letter code, copy the `translation.json` file from the `en` folder, and start translating.

- [x] Dutch
- [x] English
- [ ] French
- [ ] German
- [ ] Italian
- [ ] Spanish
- [ ] Portuguese
- [ ] Polish
- [ ] Arabic

cedricve, 2022-09-11:
Dutch and English submitted by @cedricve

cedricve, 2022-10-11:
Thank you @robdsource amazing work!

cedricve, 2022-10-20:
Thank you @TeeTeufel! Well done!!

cedricve, 2023-03-17:
Thank you @jeffersonGlemos for Portuguese and @kododake for Japanese

atomic-fart, 2023-06-13:
english

fadkeabhi, 2023-10-03:
Hey @cedricve  I would like to work on Hindi translations.

cedricve, 2023-10-22:
Italian by @LeoSpyke thank you so much!!

ghosty2004, 2024-06-25:
Hi @cedricve . I would like to translate to romanian.

cedricve, 2024-06-26:
> We are currently translating the Kerberos Agent UI. Find a list of the translations we are currently focusing on, feel free to add more languages if you are willing to contribute.
>
> Languages can be [...]

@ghosty2004 That's awesome thank you! Please follow the above instruction to understand what is required to provide translations.

cedricve, 2024-07-04:
Thank you so much @ghosty2004  we've added you to the list of contributors, well done!

mamounjamous, 2024-07-04:
Hi @cedricve, I would like to translate to Arabic.

thanhtantran, 2025-02-27:
I have created request to add Vietnamese locale