kerberos-io/agent #51: Installed Kerberos on RPI4 using Docker (Machinery Not Running Issue)

russmcg, 2022-09-16:
I followed this guide on the Kerberos.io documentation, and for some reason the Web is Running, but the Machinery is Not Running. 

https://cedricverst.medium.com/kerberos-io-on-docker-19b25fa7efd1

I am pulling in a RTSP stream, that works over VLC so I know the connection is good there. 

Curious how I go about troubleshooting the Machinery, and how to start it?

I also don't know where to find the Kerberos logs to troubleshoot.

Any help would be greatly appreciated.

cedricve, 2022-09-16:
Which agent are your running, the old kerberos/kerberos or kerberos/agent (which this is the repo)

Kind regards,
Verstraeten Cédric

russmcg, 2022-09-16:
CONTAINER ID   IMAGE               COMMAND          CREATED        STATUS        PORTS                                                                              NAMES
48d72cbbb06a   kerberos/kerberos   "bash /run.sh"   16 hours ago   Up 16 hours   0.0.0.0:8889->8889/tcp, :::8889->8889/tcp, 0.0.0.0:8080->80/tcp, :::8080->80/tcp   rustycam

russmcg, 2022-09-16:
should I have pulled kerberos/agent instead of following that tutorial I linked?

russmcg, 2022-09-16:
So update .. I removed the kerberos/kerberos docker container, and added the kerberos/agent container 

I now have the system up and running, however it still isn't pulling in my stream, which according to VLC is an H264 stream. 

Do I simply need to add and save the RTSP link or do I need to reboot or something for the changes to take effect?

cedricve, 2022-09-16:
so did you configure the new agent through the web ui?

russmcg, 2022-09-16:
Yes I imported my Wyze cam RTSP link, and it isn't showing the stream even though the stream works using VLC. Because of this I know the link should work, but there must be some incompatibility somewhere?

Any ideas?

russmcg, 2022-09-16:
This is the format of the link I'm using as delivered from the Wyze V3 RTSP firmware

rtsp://user:pass@192.168.87.202/live

cedricve, 2022-09-22:
@russmcg in latest version, we added a verify connection button, can you try that release?

Regards
Cedric

russmcg, 2022-09-22:
in the latest Kerberos/Agent version, correct? 

I just installed using 

docker run -p 8080:8080 --name rustycam -d kerberos/agent:latest

and I am not seeing a verify connection button

cedricve, 2023-02-13:
Is this still applicable with latest version @russmcg ?