kerberos-io/agent #52: Installed using Advanced Install (now unable to uninstall)

russmcg, 2022-09-22:
It looks I am having issues related to the advanced install procedure, that is still serving the kerberos agent application through port 8080. I am not sure how to uninstall this, and what service is serving the kerberos agent application. 

I followed the installation procedure backwards, and uninstalled and rm -rf a bunch of created directories during the advanced installation process, but for some reason it is still serving the kerberos agent login at https://IP:8080. I tried systemctl stop nginx and that didn't work. It is still serving the kerberos service. 

Curious how one might troubleshoot this?

olokos, 2022-09-24:
Services on debian are usually in `/etc/systemd/system/*` also rarely could be in `/etc/systemd/system/*`

cedricve, 2022-09-24:
this is related to the old agent, this repo is for the new agent @russmcg

closed 2022-12-12 by cedricve