kerberos-io/agent #57: Recordings saved with no file type

bradjwilson, 2022-11-28:
### Problem
I cannot view recordings in Kerberos Agent.

I get this error:
[image]

After downloading the recording, it is saved as "video" with no file extension.

I can see and watch the recordings just fine in my Kerberos Hub
### Troubleshooting Done
- Updating docker image kerberos/agent:latest.
- Removed passing of volumes from host to container. 

### Host Specs

- Ubuntu 20.04 VM

bradjwilson, 2022-11-28:
Seems to only happen when viewing agents in firefox.

bradjwilson, 2022-11-28:
I cannot find any extensions that help with embedded video playback in firefox.

It happens on both Ubuntu and macOS.

I tried running firefox in safe mode and doing clean installs, and the problem persists.

yllekz, 2022-11-29:
Same issue. Broken in Firefox. Needs to work in Firefox.

skoobasteeve, 2022-12-10:
Confirmed not working in Firefox, but I also wasn't able to play it in Chromium. Player loaded but no video.

[image]

My downloaded videos all save as "Video" with no file extension, could be part of the issue?

cedricve, 2022-12-10:
@skoobasteeve could be that the video is still recording, and therefore not playable. Can you confirm you could review older footages in chrome. We plan a bigger release next week with some additional features and capabilities. We will tackle the firefox issue. Will come back on this issue!

skoobasteeve, 2022-12-17:
@cedricve I can confirm that on Chromium it's not letting me play back completed recordings.

skoobasteeve, 2022-12-22:
@cedricve Digging into it a little more, seems that it works fine in Google Chrome but not Chromium.

cedricve, 2022-12-22:
thank you @skoobasteeve  for additional feedback. We will redesign the media page indeed, the idea is that you will be able filter (like the old agent) and we load the first frame of the recording (=thumbnail). I believe something is going wrong with the content-type, I'll provide a fix asap.

cedricve, 2022-12-27:
Looks like it fixed thanks to this commit: https://github.com/kerberos-io/agent/commit/9ec52d445dfe4a7150c0cf7c63bb9040a22565b8. Closing please reopen if not the case for you.

Pull latest container version `docker pull kerberos/agent:latest` and re-run the container.

closed 2022-12-27 by cedricve