kerberos-io/agent #59: Is my camera working? An overview of tested cameras.

cedricve, 2022-12-07:
New users, or existing community members, wonder if their old/new camera is expected to work with the Kerberos Agent. As reported on a previous issue kerberos-io/documentation#40, we want to provide a never-complete list of tested and approved cameras that work with the Kerberos Agent.

Call to action: If you have tested Kerberos Agent with your camera (with or without the sidecar for a USB camera), then feel free to add the camera name and model as a comment to this list. We will then update and maintain the approved camera list, so other community members can benefit from this, without buying or leveraging cameras that are not supported (or not yet supported).

- [x] Hikvision DS-2DE4415IW-DE

cedricve, 2022-12-26:
added CosterHD + sorted the list

olokos, 2023-01-15:
I can confirm that the Raspberry Pi HQ Camera / IMX477R works with the new Agent approach to cameras.

If You want to use non-default resolution then `mode` has to be set according to the mode used by the camera.

The only issue I see with the new approach is:
[image]
(Censored local IP for privacy)

Which was mentioned here as solved
https://github.com/aler9/rtsp-simple-server/issues/1133

But it still happens, with that said both Agent and VLC display the stream as they should, despite the error!
I cannot see any issues with the stream or any performance downgrades, all seem to run great, including some of the camera parameters that I used!

milankragujevic, 2023-02-03:
Cantonk IPR25FE500 - https://www.cantonk.com/product/detail/594.html

Confirmed working as basic rtsp:// stream.

nudzik, 2023-02-05:
Tp-link Tapo C310 and eufy security Solo IndoorCam C24 are working very well.

Hooch76, 2023-03-13:
Wansview W2 works fine.

cedricve, 2023-05-07:
Tried out a Reolink E1 PRO, but doesn't work as expected the RTSP stream delivered (you have 2, fluent and clear) shows artificats. It's also not reliable, even with VLC it doesn't open continuously.

I was able to connect a Kerberos Agent to the stream but most of the time it was showing encoding issues, at some points it was properly recording, but disconnected after a while.

- rtsp://xx:xx@192.168.1.xxx:554/h264Preview_01_main
- rtsp://xx:xx@192.168.1.xxx:554/h264Preview_01_sub

cedricve, 2023-05-09:
Tested, Tapo C200 works well!

cedricve, 2023-06-01:
We have verified the Reolink TrackMix PoE, as expected it gives some inconsistent behaviour. The camera comes with the two channels, for which the first channel is a wide-angle.

- The wide angle main stream doesn't work as it's a H265 stream.
- The wide angle secondary stream does work as it's a H264 stream.

In contrast to the Reolink E1 pro, this camera does give a stable secondary stream both for live streaming (WEBRTC) as recording.

We have validated the PTZ functionality, and the PTZ works properly. Please note that you'll need to enable both the RTSP and ONVIF functionality in the Reolink admin panel before you can open the stream or leverage ONVIF functions.

cedricve, 2023-11-06:
We tested Hanwha cameras, and RTSP seams to be working fine after fixing #105. ONVIF is not yet working, figuring out the root cause. Special thanks to @0xhanh

marcosr25, 2024-01-09:
One question, have you connected your cameras locally? I want to connect them externally through the public IP of the router (modem). I have already opened ports 2020 and 554 with port forwarding to 8120, 8121=2020 and 554=554 with for my tp-link tapo c310 and c320ws cameras, now I have configured the accounts of both cameras for the Onvif protocol. From the panel I add the configuration and it tells me that ONVIF settings are successfully verified. and when you click save Your configuration has been updated successfully. but I can't view them. Isn't it possible this way? thanks greetings

RickAndTired, 2024-07-07:
Amcrest IP4M-1041
RTSP works well
ONVIF does not work for me (could be a user issue)

cedricve, 2024-07-07:
Thank you for reporting @RickAndTired

cedricve, 2025-04-23:
Added Reolink RLC-1240a

cedricve, 2025-04-23:
> Tried out a [Reolink E1 PRO](https://reolink.com/product/e1-pro/?attribute_pa_version=1-pack-smart-detection-black), but doesn't work as expected the RTSP stream delivered (you have 2, fluent and clear) shows artifacts. It's also not reliable, even with VLC it doesn't open continuously.
> 
> I was able to connect a Kerberos Agent to the stream but most of the time it was showing encoding issues, at some points it was properly recording, but disconnected after a while.
> 
> * rtsp://xx:[xx@192.168.1.xxx](mailto:xx@192.168.1.xxx):554/h264Preview_01_main
> * rtsp://xx:[xx@192.168.1.xxx](mailto:xx@192.168.1.xxx):554/h264Preview_01_sub
> 
> [image]

Might have been fixed with latest release of Agent.

cedricve, 2025-09-25:
I'm not aware if someone tested that, happy to add it to the list after testing. Does it has a H264/H265 RTSP stream?

JV1260, 2025-12-05:
Reolink RLC-823A seems to be working but no PTZ controls for the camera or I just haven't found them yet.