kerberos-io/agent #61: RTSP H265 support

bigntallmike, 2022-12-16:
I'm still using the Kerberos.IO Machinery interface for my video cameras because it supports h265 streams and agent does not yet. With more and more cameras going to 4k with h265, when can we expect support in agent for h265 streams?

cedricve, 2022-12-16:
Yes that will definitely come @bigntallmike. Currently we are relying on our own version of joy4 (https://github.com/kerberos-io/joy4). This only supports PCM for audio and H264 for video as we speak, but the idea is to implement other audio and video codecs such as H265.

We are not sure if we will get rid of this external library, or we will integrate it directly into the agent repository. As it shows some artefacts with memory allocation which are hard to track. Any help from the community is more then welcome!

bigntallmike, 2023-01-06:
The only way I usually interface with RTSP streams directly is with ffmpeg itself and I lack any familiarity with golang or I'd have a look. Good luck. 
See also very old ffmpeg library example: https://stackoverflow.com/a/11054652/853220