kerberos-io/agent #63: Unable to update polygon-selection with mounted config.json

instantdreams, 2022-12-21:
# Configuration Overview

## Docker Versions
```
$ docker --version
Docker version 20.10.21, build baeda1f
$ docker compose version
Docker Compose version v2.12.2
```

## Preparation Steps
```
mkdir /srv/kerberos-agent /srv/kerberos-agent/garage-south/ /srv/kerberos-agent/garage-south/config
wget https://raw.githubusercontent.com/kerberos-io/agent/master/machinery/data/config/config.json -O /srv/kerberos-agent/garage-south/config
sudo chown 1000:1000 /srv/kerberos-agent -R
sudo chmod 777 /srv/kerberos-agent -R
mkdir /media/nvr/kerberos/garage-south /media/nvr/kerberos/garage-south/recordings /media/nvr/kerberos/garage-south/snapshots
sudo chown 1000:1000 /media/nvr/kerberos -R
sudo chmod 777 /media/nvr/kerberos -R
```

## docker-compose.yml
```
version: "3.9"
networks:
  kerberos-agent-net:
    driver: bridge
    name: kerberos-agent-net
services:
  ka-garage-south:
    image: "kerberos/agent:latest"
    container_name: ka-garage-south
    hostname: ${HOSTNAME}-ka-garage-south
    networks:
      - kerberos-agent-net
    ports:
      - "80:80"
    environment:
      - AGENT_USERNAME=${AGENT_USERNAME}
      - AGENT_PASSWORD=${AGENT_PASSWORD}
    volumes:
      - ${APPDATA}/garage-south/config:/home/agent/data/config:rw
      - ${RECORDINGS}/garage-south/recordings:/home/agent/data/recordings:rw
      - ${RECORDINGS}/garage-south/snapshots:/home/agent/data/snapshots:rw
    restart: unless-stopped
```

## .env
```
# Directory locations
APPDATA=/srv/kerberos-agent
RECORDINGS=/media/nvr/kerberos
# Container specifics
AGENT_USERNAME=[username]
AGENT_PASSWORD=[password]
HOSTNAME=[hostname]
```

## config.json
```
{
        "type": "",
        "key": "",
        "name": "garage-south",
        "time": "true",
        "offline": "false",
        "auto_clean": "true",
        "max_directory_size": 102400,
        "timezone": "America/Edmonton",
        "capture": {
[... 287 lines cut ...]
                                                "x": 1909.6776713709678,
                                                "y": 1073.4195438508063
                                        },
                                        {
                                                "x": 530.5807711693549,
                                                "y": 1073.4195438508063
                                        }
                                ]
                        }
                ]
        },
```

## Browser Details
* Google Chrome 108.0.5359.125 (64-bit)
* Mozilla Firefox 108.0.1 (64-bit)

## Results Summary
Test 5: ✔️ Settings
Test 7: ✔️ Recording
Test 9: ❌ Polygon
Test 11: ✔️ Rectangle

## Conclusion
I tried this a few times on Firefox and Chrome, but nothing saves.

cedricve, 2022-12-21:
Thanks again for the detailed logs! I also noticed your camera is restarting. Is this happening continuously?
    
    ka-garage-south  | {"level":"info","msg":"HandleRecordStream: Start continuous recording ","time":"2022-12-21T09:35:09-07:00"}
    ka-garage-south  | {"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2022-12-21T09:35:10-07:00"}
    ka-garage-south  | {"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2022-12-21T09:35:15-07:00"}
    ka-garage-south  | {"level":"info","msg":"Main: Restarting machinery.","time":"2022-12-21T09:35:15-07:00"}
    ka-garage-south  | {"level":"error","msg":"HandleStream: read tcp 192.168.32.2:43006-\u003e192.168.1.83:554: use of closed network connection","time":"2022-12-21T09:35:16-07:00"}
    ka-garage-south  | {"level":"error","msg":"HandleSubStream: read tcp 192.168.32.2:43018-\u003e192.168.1.83:554: use of closed network connection","time":"2022-12-21T09:35:16-07:00"}
    ka-garage-south  | {"level":"info","msg":"WriteToTrack: stop writing to track.","time":"2022-12-21T09:35:16-07:00"}

cedricve, 2023-04-12:
Not sure if still relevant @instantdreams, but it looks like it might be linked to user permissions: https://github.com/kerberos-io/agent/issues/94#issuecomment-1504325778.