kerberos-io/agent #65: Url to access ui

paucazou, 2022-12-28:
Hi,
thanks for your wonderful product.
I've tried to install Kerberos Agent through docker and everything seems to work until I try to get to the interface. The client (firefox in my case) says it's unable to connect to the server.
I've installed the agent on a raspberry pi Zero with the latest Raspberry Pi OS lite version. I can access the pi through ssh.
the command I entered to launch the service is:
`sudo docker run -p 80:80 --name kerberos0 -d kerberos/agent:latest`
Maybe I must add that I've encountered this warning, maybe related:
`WARNING: The requested image's platform (linux/arm/v7) does not match the detected host platform (linux/arm/v6) and no specific platform was requested`

Thanks in advance.

cedricve, 2022-12-28:
hey @paucazou, thanks! That's a bummer for the moment we do not have a armv6 compatible image. However this is on our roadmap and will be released in the next weeks. For the moment its recommended to run on a Raspberry Pi 2 or higher.

paucazou, 2022-12-28:
Ok, thanks for your (really quick!) answer. I'll try again then.

cedricve, 2022-12-28:
no worries and welcome @paucazou, I just renamed the issue to make the issue more obvious

cedricve, 2023-02-13:
This is now available, we also have static binaries now -> https://github.com/kerberos-io/agent/releases

closed 2023-02-13 by cedricve