kerberos-io/agent #66: Camera not showing rPi 4B 4GB RAM

kwolk, 2023-01-01:
The official Raspberry Pi Camera Module 2 is not showing on a Raspberry Pi 4B model its 4GB RAM, running the latest updated release of Bullseye.

yllekz, 2023-01-03:
Have you gone through the documentation on [rtsp-simple-server](https://github.com/aler9/rtsp-simple-server/)? There's a few gotchas to consider for the rPi camera. Works fine for me but it took a bunch of trial-and-error to get it configured just right. Read the docs carefully on that page and it should get you set.

cedricve, 2023-01-03:
@yllekz we have created a dedicated image for that now (kerberos/rtsp-simple-server:latest-rpi). As described here, https://github.com/kerberos-io/camera-to-rtsp#raspberry-pi-camera, this new image now ships the correct libcamera, and avoids attaching `/usr` into the container.

cedricve, 2023-02-13:
Let me know if anymore question on this topic! ;)