kerberos-io/agent #75: Docker instance won't connect to IP cam

ryokimball, 2023-02-01:
I feel bad for asking because it seems obvious that the product works but I am stumped on the very first step in using it.

I have a WyzeCamV2 running [OpenMiko](https://github.com/openmiko/openmiko) firmware, which supports (and I have configured for) H264 RTSP streaming. I can view the video stream through VLC with the url rtsp://192.168.1.199:8554/video3_unicast. I am executing the command docker run -p 8081:80 --name mycamera -e AGENT_CAPTURE_IPCAMERA_RTSP="rtsp://192.168.1.199:8554/video3_unicast" kerberos/agent:latest on my host. The agent seems to run fine and I can connect via browser. With our without AGENT_CAPTURE_IPCAMERA_RTSP... set, I cannot add the RTSP stream; no cameras show attached and when I try to connect one via dashboard->settings, it just spins. The logs aren't really helpful (see below).
```

{"level":"info","msg":"Main: updated unique key for agent to: GE5WRJKLBS6N731MS9BYNUD6OEGSBQ","time":"2023-01-26T03:23:39+01:00"}
[GIN-debug] [WARNING] Creating an Engine instance with the Logger and Recovery middleware already attached.

[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:   export GIN_MODE=release
 - using code:  gin.SetMode(gin.ReleaseMode)

[GIN-debug] GET    /debug/pprof/             --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/cmdline      --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/profile      --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] POST   /debug/pprof/symbol       --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/symbol       --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/trace        --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/allocs       --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/block        --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/goroutine    --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/heap         --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/mutex        --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/threadcreate --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /swagger/*any             --> github.com/swaggo/gin-swagger.CustomWrapHandler.func1 (4 handlers)
[GIN-debug] GET    /ws                       --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func1 (4 handlers)
[GIN-debug] GET    /config                   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func2 (4 handlers)
[GIN-debug] POST   /config                   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func3 (4 handlers)
[GIN-debug] POST   /api/login                --> github.com/appleboy/gin-jwt/v2.(*GinJWTMiddleware).LoginHandler-fm (4 handlers)
[GIN-debug] GET    /api/dashboard            --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func4 (4 handlers)
[GIN-debug] POST   /api/latest-events        --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func5 (4 handlers)
[GIN-debug] GET    /api/days                 --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func6 (4 handlers)
[GIN-debug] GET    /api/config               --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func7 (4 handlers)
[GIN-debug] POST   /api/config               --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func8 (4 handlers)
[GIN-debug] GET    /api/restart              --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func9 (4 handlers)
[GIN-debug] GET    /api/stop                 --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func10 (4 handlers)
[GIN-debug] POST   /api/hub/verify           --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func11 (4 handlers)
[GIN-debug] POST   /api/persistence/verify   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func12 (4 handlers)
[GIN-debug] GET    /api/stream               --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func13 (4 handlers)
[GIN-debug] POST   /api/camera/onvif/login   --> github.com/kerberos-io/agent/machinery/src/routers/http.LoginToOnvif (4 handlers)
[GIN-debug] POST   /api/camera/onvif/capabilities --> github.com/kerberos-io/agent/machinery/src/routers/http.GetOnvifCapabilities (4 handlers)
[GIN-debug] POST   /api/camera/onvif/pantilt --> github.com/kerberos-io/agent/machinery/src/routers/http.DoOnvifPanTilt (4 handlers)
[GIN-debug] POST   /api/camera/onvif/zoom    --> github.com/kerberos-io/agent/machinery/src/routers/http.DoOnvifZoom (4 handlers)
[GIN-debug] POST   /api/camera/verify/:streamType --> github.com/kerberos-io/agent/machinery/src/capture.VerifyCamera (4 handlers)
[GIN-debug] GET    /file/*filepath           --> github.com/kerberos-io/agent/machinery/src/routers/http.Files (9 handlers)
[GIN-debug] [WARNING] You trusted all proxies, this is NOT safe. We recommend you to set a value.
Please check https://pkg.go.dev/github.com/gin-gonic/gin#readme-don-t-trust-all-proxies for details.
[GIN-debug] Listening and serving HTTP on :80
{"level":"info","msg":"RunAgent: opening RTSP stream","time":"2023-01-26T03:23:39+01:00"}
{"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2023-01-26T03:23:39+01:00"}
{"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2023-01-26T03:23:44+01:00"}
[GIN] 2023/01/26 - 02:23:45 | 200 |     795.269µs |   192.168.1.103 | GET      "/api/dashboard"
{"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2023-01-26T03:23:49+01:00"}
{"level":"info","msg":"Main: Restarting machinery.","time":"2023-01-26T03:23:49+01:00"}
[GIN] 2023/01/26 - 02:23:50 | 200 |     142.375µs |   192.168.1.103 | GET      "/api/dashboard"
{"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2023-01-26T03:23:56+01:00"}
[GIN] 2023/01/26 - 02:24:00 | 200 |      389.26µs |   192.168.1.103 | GET      "/api/dashboard"
{"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2023-01-26T03:24:01+01:00"}
[GIN] 2023/01/26 - 02:24:05 | 200 |     438.809µs |   192.168.1.103 | GET      "/api/dashboard"
{"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2023-01-26T03:24:06+01:00"}
{"level":"info","msg":"Main: Restarting machinery.","time":"2023-01-26T03:24:06+01:00"}
[GIN] 2023/01/26 - 02:24:09 | 200 |     279.666µs |       127.0.0.1 | GET      "/"
[GIN] 2023/01/26 - 02:24:10 | 200 |     516.883µs |   192.168.1.103 | GET      "/api/dashboard"
{"level":"info","msg":"ForwardSDStream: stop sending streaming over websocket","time":"2023-01-26T03:24:29+01:00"}
[GIN] 2023/01/26 - 02:24:31 | 200 |     680.807µs |   192.168.1.103 | GET      "/api/config"
[GIN] 2023/01/26 - 02:24:32 | 200 |     574.596µs |   192.168.1.103 | GET      "/api/dashboard"

```

cedricve, 2023-02-01:
Hey @ryokimball looks like a stream issue.  `rtsp://192.168.1.199:8554/video3_unicast` does your camera also ship other RTSP urls? What are the current video settings of your camera (FPS, Codec, etc), could you share some more details?

ryokimball, 2023-02-01:
I believe this is the only RTSP stream (there are two other HTTP streams). 
FPS: 25
Codec: PT_H264
[Here are the settings for the video streams from the firmware](https://github.com/openmiko/ingenic_videocap/blob/master/settings.json) (admittedly I don't understand all/most of it) 

I found a "turn your Android phone into an IP camera" app that supports RTSP and tested the docker image with that instead and it works; so the problem is definitely with my stream. (I won't be offended if the ticket is closed as "not our problem" but I appreciate any help moving forward)

cedricve, 2023-02-01:
@ryokimball no worries, we would like to solve your issue, as it can also be beneficial for other people so do not mind asking us. So from my understanding you are able to modify that `json` file to tweak your RTSP stream?

cedricve, 2023-02-01:
And if yes could change this settings, and modify it to `PIX_FMT_YUV420P` instead.
https://github.com/openmiko/ingenic_videocap/blob/master/settings.json#L16

cedricve, 2023-02-13:
did this help @ryokimball ?

ryokimball, 2023-02-15:
Sorry for the slow response; that one change to the config did not fix it and I have not had time to try other changes

j-broome, 2023-06-10:
Any update on this issue? I am running the RTSP firmware on the Wyze Cam v3 and can also view in VLC, but I get the following in the agent: "Something went wrong while verifying the camera settings :rtsp: Type=466673 unsupported".

cedricve, 2023-06-10:
Hello @j-broome, can you share the codec settings of your Wyze cam3? We recommended H264 for video and PCM or AAC for audio.

j-broome, 2023-06-10:
Where do I find this? I can't see the .bin file that wyze gives you to enable the rtps setup. Like I said I can pull it up in VNC with rtsp://user:pass@xxx.xxx.xxx.xxx/live. Just tested with a Shinobi container and had no issue.

cedricve, 2023-06-10:
Can you open the RTSP stream with VLC, an take a snapshot of the media information? This will show the codecs information.

j-broome, 2023-06-10:
[image]

j-broome, 2023-06-10:
[image]