kerberos-io/agent #77: Tapo Camera Audio Support

harpyeagle, 2023-02-13:
Hi @cedricve 

I am using the agent to record birds in a number of nestboxs. I am also signed up to the Hub as the analytics is a great addition to quickly see the activity during the day. The issue I have is no audio, the rtsp stream contains audio and I have verified this on VLC player but the agent / hub are not picking this up. Is there a setting I am missing to enable audio on the recording.

Thanks John

cedricve, 2023-02-13:
hey there @harpyeagle which audio codec do you have selected for your Tapo Camera?

harpyeagle, 2023-02-13:
Hi @cedricve 

Based on VLC it is it is PCM ALAW

[image]

cedricve, 2023-02-13:
Is there an option to modify to PCM_MULAW?

cedricve, 2023-02-13:
Apologies modify to AAC (this is required for MP4). We are still working on audio for livestream, but you can expect this enhancement soon. AAC audio is already supported for recordings.

cedricve, 2023-05-27:
PCM_ALAW is now supported in live streaming, still needs transcoding to AAC to work in MP4.

cedricve, 2023-12-29:
Bi-directional audio is now added, but it lacks support (incoming) for Tapo cameras as they don't support the ONVIF backchannel method. To transfer audio through the livestream on Kerberos Hub, the camera stream needs to support (and has enabled) PCM_ULAW. To record audio in a MP4 file (recording) the audio needs to be set to AAC.

At the moment we are working on transcoding from AAC -> PCM_ULAW and AAC -> PCM_ULAW to make sure we can support both scenarios.

closed 2023-12-29 by cedricve