kerberos-io/agent #81: Can't record anything

Hooch76, 2023-03-08:
Hi,

I just deployed the Kerberos agent docker container but can't record anything ("Continous recording" on or off doesn't matter).

This is what the log tells me

```023-03-08T23:41:02+01:00 INF msg=ProcessMotion: Motion detection enabled.
2023-03-08T23:41:02+01:00 INF msg=HandleRecordStream: disabled, we will not record anything.
2023-03-08T23:41:02+01:00 INF msg=HandleLiveStreamHD: Waiting for peer connections.
2023-03-08T23:41:02+01:00 INF msg=WriteToTrack: listing codecs.
2023-03-08T23:41:02+01:00 INF msg=WriteToTrack: codec - H264 found.
2023-03-08T23:41:02+01:00 INF msg=H264
2023-03-08T23:41:02+01:00 INF msg=WriteToTrack: codec - AAC found.
2023-03-08T23:41:02+01:00 INF msg=AAC
2023-03-08T23:41:02+01:00 INF msg=WriteToTrack: not using a transcoder.
2023-03-08T23:41:02+01:00 ERR msg=HandleHeartBeat: (400) Something went wrong while sending to Kerberos Hub.
2023-03-08T23:41:02+01:00 INF msg=HandleStream: packet size 13123
```

What causes "HandleRecordStream: disabled, we will not record anything."?

I can see a picture of the camera ("Live view") on the Dashboard and "Verify Connection" (Settings > Camera) results in "Camera settings are successfully verified."

Thanks,
Hooch

cedricve, 2023-03-09:
hey @Hooch76 can you share with us the docker image version you have deployed? This is unexpected behaviour, there are environment variables to disable specific parts of the agent but they should be specified explicitly.

Hooch76, 2023-03-09:
This is part of my Dockerfile:

```
  kerberos-agent:
    image: "kerberos/agent:latest"
```

cedricve, 2023-03-09:
Are you injecting a config file? If yes, can you attach it to this issue. If no can you provide a fixed release, instead of the latest release. You'll find all releases here: https://hub.docker.com/r/kerberos/agent/tags.

    kerberos-agent:
        image: "kerberos/agent:c4cda0a"

Hooch76, 2023-03-09:
I am using the config file that is mentioned in the docs: https://github.com/kerberos-io/agent/tree/master/deployments/docker#attaching-volumes --> https://raw.githubusercontent.com/kerberos-io/agent/master/machinery/data/config/config.json

cedricve, 2023-03-09:
Can you please upgrade to latest version (the one I have commented above)?

Hooch76, 2023-03-09:
Upgraded but it did not change anything:

```
2023-03-09T12:01:08+01:00 INF msg=Main: Restarting machinery.
2023-03-09T12:01:08+01:00 INF msg=RunAgent: waiting 3 seconds to make sure everything is properly closed.
2023-03-09T12:01:11+01:00 INF msg=Successfully Opened config.json from opensource
2023-03-09T12:01:11+01:00 INF msg=RunAgent: opening RTSP stream
[{video H264 0 90000 trackID=0 96 0 [] [[103 100 0 30 172 232 10 2 255 149] [104 238 60 176]] [] [] [] 96 0 0} {audio AAC 0 16000 trackID=1 97 0 [20 8] [] [] [] [] 97 13 3}]
[GIN] 2023/03/09 - 11:01:12 | 200 |     458.364µs | 192.168.178.202 | GET      "/api/dashboard"
2023-03-09T12:01:12+01:00 INF msg=RunAgent: opened RTSP stream
2023-03-09T12:01:12+01:00 INF msg=RunAgent: camera settings did not change, keeping decoder
2023-03-09T12:01:12+01:00 INF msg=RunAgent: SetMaxGopCount was set with: 6
2023-03-09T12:01:12+01:00 INF msg=ConfigureMQTT: Set broker uri 
2023-03-09T12:01:15+01:00 INF msg=ProcessMotion: Motion detection enabled.
2023-03-09T12:01:15+01:00 INF msg=HandleRecordStream: disabled, we will not record anything.
2023-03-09T12:01:15+01:00 INF msg=HandleLiveStreamHD: Waiting for peer connections.
2023-03-09T12:01:15+01:00 INF msg=WriteToTrack: listing codecs.
2023-03-09T12:01:15+01:00 INF msg=WriteToTrack: codec - H264 found.
2023-03-09T12:01:15+01:00 INF msg=H264
2023-03-09T12:01:15+01:00 INF msg=WriteToTrack: codec - AAC found.
2023-03-09T12:01:15+01:00 INF msg=AAC
```

cedricve, 2023-03-09:
Can you run the following:

    docker run -p 80:80 -e AGENT_CAPTURE_IPCAMERA_RTSP=rtsp://fake.kerberos.io/stream  kerberos/agent

Hooch76, 2023-03-09:
This does work:

```
{"level":"info","msg":"Successfully Opened config.json from opensource","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"RunAgent: opening RTSP stream","time":"2023-03-09T13:34:04+01:00"}
[{video H264 0 90000 trackID=0 96 0 [] [[103 100 0 41 172 27 26 128 120 2 39 229 192 91 128 128 128 160 0 0 125 0 0 29 76 29 12 0 5 245 224 0 1 220 214 87 121 113 161 128 0 190 188 0 0 59 154 202 239 46 20] [104 238 56 48]] [] [] [] 96 0 0}]
{"level":"info","msg":"RunAgent: opened RTSP stream","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"RunAgent: camera settings did not change, keeping decoder","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"RunAgent: SetMaxGopCount was set with: 11","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"ConfigureMQTT: Set broker uri tcp://mqtt.kerberos.io:1883","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"ProcessMotion: Motion detection enabled.","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"HandleLiveStreamHD: Waiting for peer connections.","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"WriteToTrack: listing codecs.","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"WriteToTrack: codec - H264 found.","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"H264","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"WriteToTrack: not using a transcoder.","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"HandleRecordStream: Start motion based recording ","time":"2023-03-09T13:34:04+01:00"}
{"level":"error","msg":"HandleHeartBeat: (400) Something went wrong while sending to Kerberos Hub.","time":"2023-03-09T13:34:04+01:00"}
{"level":"info","msg":"HandleStream: packet size 205924","time":"2023-03-09T13:34:05+01:00"}
[GIN] 2023/03/09 - 12:34:07 | 200 |     504.769µs |  192.168.178.86 | GET      "/api/dashboard"
{"level":"info","msg":"HandleStream: packet size 205766","time":"2023-03-09T13:34:07+01:00"}
{"level":"info","msg":"ControlAgent: Number of packets read 44","time":"2023-03-09T13:34:09+01:00"}
{"level":"info","msg":"HandleStream: packet size 208142","time":"2023-03-09T13:34:09+01:00"}
{"level":"info","msg":"HandleStream: packet size 206550","time":"2023-03-09T13:34:12+01:00"}
{"level":"info","msg":"HandleRecordStream: Recording started","time":"2023-03-09T13:34:12+01:00"}
{"level":"info","msg":"HandleRecordStream: composing recording","time":"2023-03-09T13:34:12+01:00"}
{"level":"info","msg":"HandleRecordStream: write header","time":"2023-03-09T13:34:12+01:00"}
{"level":"info","msg":"HandleRecordStream: write frames","time":"2023-03-09T13:34:12+01:00"}
{"level":"info","msg":"HandleRecordStream: Synced file: 1678365243_6-967003_agent_200-200-400-400_13716_769.mp4","time":"2023-03-09T13:34:12+01:00"}
{"level":"info","msg":"HandleRecordStream: Synced file: 1678365243_6-967003_agent_200-200-400-400_13716_769.mp4","time":"2023-03-09T13:34:12+01:00"}
{"level":"info","msg":"HandleRecordStream: Synced file: 1678365243_6-967003_agent_200-200-400-400_13716_769.mp4","time":"2023-03-09T13:34:12+01:00"}
{"level":"info","msg":"HandleRecordStream: Synced file: 1678365243_6-967003_agent_200-200-400-400_13716_769.mp4","time":"2023-03-09T13:34:12+01:00"}
```

Hooch76, 2023-03-09:
Just solved it!

I examined the config.json file that's present when running 

> docker run -p 80:80 -e AGENT_CAPTURE_IPCAMERA_RTSP=rtsp://fake.kerberos.io/stream  kerberos/agent

and compared it with the config file https://github.com/kerberos-io/agent/blob/master/machinery/data/config/config.json that I downloaded yesterday. (It was changed 11 hours ago; so most likely _after_ I downloaded it).

The copy I was using contained

> "recording": "false"

Changing it to

> "recording": "true"

solved the problem.

cedricve, 2023-03-12:
Lovely!

closed 2023-03-12 by cedricve