kerberos-io/agent #84: mqtt message - Double slash

Hooch76, 2023-03-13:
Hi,

I can see that the Kerberos agent

cedricve, 2023-03-17:
This is related to the upload issue with missing credentials! Thanks for reporting. If no credentials are set, no outgoing MQTT session should be made to Hub as well!

cedricve, 2023-03-19:
Hey @Hooch76, I just had a look and it should not open any outgoing MQTT missages if the hubkey is not set. https://github.com/kerberos-io/agent/blob/master/machinery/src/routers/mqtt/main.go#L62

Can you validate your config, and the version of your Kerberos Agent?

Hooch76, 2023-03-20:
I am using d9694ac and a local MQTT server:

> Sending PUBLISH to <my_local_server>(d0, q0, r0, m0, 'kerberos//device/cam_front_door/motion', ... (6 bytes))

My local MQTT server is configured under Settings > Streaming: Broker URI, Username and Password.

cedricve, 2023-03-20:
Ok thanks for the feedback! Will verify that.

Hooch76, 2023-03-21:
Just a question: now I am using 

> image: "kerberos/agent:71cd315"

Nothing is configured in the "Kerberos Hub", "Persistence" and "STUN/TURN" sections, but I've configured MQTT (Broker URI (which points to my own MQTT server), Username and Password). 
This does no longer trigger a MQTT message when motion is detected (which is bad because I was using MQTT to kick off stuff in Home Assistant). MQTT messages were sent in https://github.com/kerberos-io/agent/releases/tag/d9694ac though.

Is this a configuration error on my side or a bug?

cedricve, 2023-03-22:
Good point! I didn't get the usecase, apologies. Now I understand the underlaying question. When bringing your own MQTT broker, without connecting to Kerberos Hub, you still want to receive motion messages on your own broker. Therefore we cannot disable the MQTT messages on motion (what we did now).

As you suggested earlier, I propose the following:

Kerberos Hub connected -> Motion -> kerberos/${hubkey}/device/${devicekey}...
No Kerberos Hub connected -> Motion -> kerberos/device/${devicekey}...

Hooch76, 2023-03-23:
> Kerberos Hub connected -> Motion -> kerberos/${hubkey}/device/${devicekey}... No Kerberos Hub connected -> Motion -> kerberos/device/${devicekey}...

Sounds good (if you were asking me ;-) )

cedricve, 2023-03-23:
fixed in latest release! https://github.com/kerberos-io/agent/blob/master/machinery/src/computervision/main.go#L167-L173

closed 2023-03-23 by cedricve

Hooch76, 2023-03-24:
Works perfectly fine. Thank you!