kerberos-io/agent #85: Disable uploads

Hooch76, 2023-03-15:
Hi, 

is there a way to disable uploading mp4 files?

> 2023-03-15T12:47:56+01:00 INF msg=UploadS3: Upload started for <xxx>.mp4
> 2023-03-15T12:47:56+01:00 ERR msg=UploadS3: Uploading Failed, Bucket name cannot be empty

There are is nothing configured in Settings -> Persistence: Kerberos Hub and Persistence. 

I am aware of the option "Offline mode", but that also disables MQTT (which is running locally) so this does not help.

Thanks.

cedricve, 2023-03-15:
> 2023-03-15T12:47:56+01:00 INF msg=UploadS3: Upload started for .mp4

Good point, we will add that to our roadmap. Indeed offline is the current work-a-around, but that might not cover all scenarios (as what you use mentioned). Thanks for the feedback!

cedricve, 2023-03-16:
We will add a disable upload option asap, (similar to offline mode) https://github.com/kerberos-io/agent/blob/master/machinery/src/cloud/Cloud.go#L53

cedricve, 2023-03-17:
I have added a fix, please pull latest release! ;) Thanks for opening this issue! https://github.com/kerberos-io/agent/commit/58c3e73f6f42ca72d78e7da2c5481a6f511b805e. Please close if sufficient.

Hooch76, 2023-03-20:
Hi,

I am using kerberos agent d9694ac and those lines are no longer appearing in the logs, so this seems to be fixed:

> INF msg=UploadS3: Upload started for .mp4
> ERR msg=UploadS3: Uploading Failed, Bucket name cannot be empty

I don't know if it is related, but now I see every 30s or so: 

> RR msg=HandleHeartBeat: (400) Something went wrong while sending to Kerberos Hub.

I've to wait for something moving in front of the camera to be able to tell if MQTT is still working though ;-)

Hooch76, 2023-03-20:
Update ... this is how it looks like now:

> INF msg=HandleRecordStream: Synced file: 1679313244_6-967003_xxyyzz_200-200-400-400_1082_769.mp4
> INF msg=HandleRecordStream: motion detected while recording. Expanding recording.
> INF msg=Received message with recording data, detected changes to save: 8999
> ERR msg=UploadS3: Uploading Failed, as no credentials found
> ERR msg=UploadS3: Uploading Failed, as no credentials found
> ERR msg=UploadS3: Uploading Failed, as no credentials found
> INF msg=HandleStream: packet size 36379
> INF msg=HandleRecordStream: Synced file: 1679313244_6-967003_xxyyzz_200-200-400-400_1082_769.mp4
> INF msg=HandleRecordStream: motion detected while recording. Expanding recording.
> INF msg=Received message with recording data, detected changes to save: 7026
> ERR msg=UploadS3: Uploading Failed, as no credentials found

cedricve, 2023-03-21:
This has been fixed by https://github.com/kerberos-io/agent/commit/71cd315142a97f56596bfe3b56a3c5aadb45d592. Thanks @Hooch76