kerberos-io/agent #87: docker deployment: cannot bind to port 80

tca22, 2023-03-17:
Hi,

I try to run kerberos-agent from scratch with docker: `docker run --name camera1 -p 41980:80 -d kerberos/agent:latest`

The container is shutdown immediately with the log entry:

`[GIN-debug] [ERROR] listen tcp :80: bind: permission denied
2023/03/17 09:25:56 listen tcp :80: bind: permission denied`

I tried `--privileged` execution as well - no success. 

As far as I understand, port 80 is used for binding even though it should be 41980. 
Do I miss something?

Thanks!

cedricve, 2023-03-17:
hello @tca22, so if you bind on port 80 it does work `-p 80:80`. What happens if you bind to port 81 `-p 81:80`.

tca22, 2023-03-17:
Hello @cedricve,
I see my title is misleading.
I would like to use a different port than 80, as 80 is already in use.
So the error-messages are different, but neither -p 80:80, -p 81:80 or e.g. -p 81980:80 work.

-p 80:80
`docker: Error response from daemon: driver failed programming external connectivity on endpoint camera1 (3d51e7a4da542bf1dfe777219d6209203716c5653918357c2061ca77d2ce3e14): Error starting userland proxy: listen tcp4 0.0.0.0:80: bind: address already in use.`

-p 81:80  or  -p 41980:80
`docker: Error response from daemon: OCI runtime create failed: container_linux.go:367: starting container process caused: process_linux.go:495: container init caused: apply apparmor profile: apparmor failed to apply profile: write /proc/self/attr/exec: no such file or directory: unknown.`

I do run Docker on Synology with a few "kerberos/kerberos" containers, but would like to use the newer "kerberos/agent" developments ...

Any thoughts?

cedricve, 2023-03-17:
Interesting! Let me check on my synology as well. Might makes sense to do so, and store your recordings in Kerberos Vault on your Synology as well.

cedricve, 2023-03-17:
@tca22 I just tried it in my own Synology, using the Docker ui, and I didn't had issues loading it. 

[image]

tca22, 2023-03-17:
Doesn't work for me, strange!

I removed the image, so I can do a fresh install from the command-line:

1) `sudo docker run -p 41980:80 --name mycamera -d kerberos/agent:latest`
container is downloaded, no error on the command-line

2) docker-ui shows that the container is stoped
the container-log attached
[mycamera.csv](https://github.com/kerberos-io/agent/files/11005238/mycamera.csv)

Well, seems to be a local issue then. Would you mind giving some directions where to start looking?
Thanks!

cedricve, 2023-03-17:
@tca22 which release are you using? Can you take a specific release? https://hub.docker.com/layers/kerberos/agent/192f78a/images/sha256-334d5416fb129e83da82435f584d02c9774f88404a813bb32aba1da73ae43bd3?context=explore

`kerberos/agent:192f78a`

tca22, 2023-03-17:
@cedricve  had `agent:latest`, tried this now `sudo docker run -p 41980:80 --name mycamera -d kerberos/agent:192f78a`, but no change, same error :-/

cedricve, 2023-03-17:
What architecture is your Synology (version) running on?

cedricve, 2023-03-17:
What `docker version` are you using?

cedricve, 2023-09-07:
@tca22  do you have an update on this ?