kerberos-io/agent #88: app.kerberos.io - Day Overview Question

harpyeagle, 2023-03-17:
@cedricve I think I know what the Day Overview is but can you explain - is it joining all recordings into one and the timeline on the right is a way to jump to each clip? ( the movie does not load for me )

Related to this, is it possible to download the number of events per day data in a csv or json  file?

Again just to say that the app.kerberos.io analytics are great for my use case of tracking activity in bird nest boxes - thanks for all the work, Im getting great value from the subscription.

John

cedricve, 2023-03-22:
Thanks @harpyeagle, can you upgrade to the latest version? And can you confirm you have the "Fragmented recording" option enabled in the "Recordings tab" on the settings page.

> Related to this, is it possible to download the number of events per day data in a csv or json file?
Interesting what is the usecase? To important in a tool such as Power BI for dashboarding? Or you would like to have this in the Kerberos Hub UI?

harpyeagle, 2023-04-10:
Hi @cedricve 

I'm running into an issue when Fragmented recording is turned on. If I download a recorded video I am no longer able to do basic edits on it in Quicktime - "Export", "Trim" etc. Is this related to the Fragmented recording setting? Its like they are read-only.

Thanks,
John

harpyeagle, 2023-04-15:
Hi @cedricve 

Just to answer my own question - if you do run into the above issue an alternative is to use an application called Compressor on the mac store.

Thanks,
John

closed 2023-12-29 by cedricve