kerberos-io/agent #89: The "recordings" tab is empty if the agent name contains an underscore "_"

AMD-NICK, 2023-03-19:
I am not familiar with GoLang or the project structure, but it seems that the issue is related to this code: https://github.com/kerberos-io/agent/blob/192f78ae78891b4d7949e63c6b84507619a6a23c/machinery/src/utils/main.go#L204

When running the agent with the environment variable `AGENT_NAME=some_name`, the "Number of days" field in the dashboard always shows 0, while the "Total recordings" field displays the correct number of recordings. However, the "Recordings" tab and "Latest events" remain empty.

Changing the agent's name from "some_name" to "some-name" fixes the issue.

AMD-NICK, 2023-03-19:
[image]

Perhaps a good solution would be to create a parsing function for the name based on a regular expression like this: `^(\d+)_(\d+-\d+)_(.+)_(\d+-\d+-\d+-\d+)_(\d+)_(\d+).mp4$`

cedricve, 2023-03-19:
thanks @AMD-NICK  you are 100% correct. This is a legacy thing, within the Kerberos Hub we expect files to be formatted using `_` to separate different variables. Therefore we don't allow agent names including a `_`, they will be replace by a `-` when using the UI, but don't do this with environment variables.

Would think it makes sense to have an implicit rename in the startup process of the agent, and throw a warning in the console log?

AMD-NICK, 2023-03-19:
Yes, I think it would be a good solution, although it sounds a little "hacky" 🙂

cedricve, 2023-03-19:
Yes I agree, and it should be done differently but we would need to rework some parts of our applications to handle that. Looks like a strange design decision in the past, but still this can be integrated. Let's leave the issue open for a reference, and future work. Thank you for your input @AMD-NICK  highly appreciated!