kerberos-io/agent #91: Static binary - Fatal error: unexpected signal during runtime execution

AntonVonDelta, 2023-04-01:
Hi! I am trying to run the static binaries but I get the following error:
**Ubuntu 22.10**

```
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x233b0 pc=0x233b0]

runtime stack:
runtime.throw({0x20e2763?, 0x4180048?})
        /usr/local/go/src/runtime/panic.go:1047 +0x5d fp=0x7ffecfd28f68 sp=0x7ffecfd28f38 pc=0x4c273d
runtime.sigpanic()
        /usr/local/go/src/runtime/signal_unix.go:819 +0x369 fp=0x7ffecfd28fb8 sp=0x7ffecfd28f68 pc=0x4d8de9

goroutine 1 [syscall, locked to thread]:
runtime.cgocall(0xf92800, 0xc00002d578)
        /usr/local/go/src/runtime/cgocall.go:158 +0x5c fp=0xc00002d550 sp=0xc00002d518 pc=0x48e43c
os/user._Cfunc_mygetpwuid_r(0x0, 0xc00049f380, 0x417e460, 0x400, 0xc0000b0228)
        _cgo_gotypes.go:176 +0x4c fp=0xc00002d578 sp=0xc00002d550 pc=0xbba24c
os/user.lookupUnixUid.func1.1(0x0, 0x3807780?, 0xc00002d610?, 0x498207?)
        /usr/local/go/src/os/user/cgo_lookup_unix.go:102 +0xa6 fp=0xc00002d5d8 sp=0xc00002d578 pc=0xbba966
os/user.lookupUnixUid.func1()
        /usr/local/go/src/os/user/cgo_lookup_unix.go:102 +0x29 fp=0xc00002d608 sp=0xc00002d5d8 pc=0xbba8a9
os/user.retryWithBuffer(0xc0000ab800, 0xc00002d6e8)
        /usr/local/go/src/os/user/cgo_lookup_unix.go:244 +0x39 fp=0xc00002d650 sp=0xc00002d608 pc=0xbbaf99
os/user.lookupUnixUid(0x0)
        /usr/local/go/src/os/user/cgo_lookup_unix.go:95 +0x10f fp=0xc00002d728 sp=0xc00002d650 pc=0xbba6af
os/user.current()
        /usr/local/go/src/os/user/cgo_lookup_unix.go:48 +0x47 fp=0xc00002d768 sp=0xc00002d728 pc=0xbba587
os/user.Current.func1()
        /usr/local/go/src/os/user/lookup.go:15 +0x17 fp=0xc00002d778 sp=0xc00002d768 pc=0xbb9fd7
sync.(*Once).doSlow(0x0?, 0xc00002d7e8?)
        /usr/local/go/src/sync/once.go:74 +0xc2 fp=0xc00002d7d8 sp=0xc00002d778 pc=0x500b82
sync.(*Once).Do(...)
        /usr/local/go/src/sync/once.go:65
os/user.Current()
        /usr/local/go/src/os/user/lookup.go:15 +0x37 fp=0xc00002d808 sp=0xc00002d7d8 pc=0xbba077
github.com/golang/glog.init.1()
        /go/pkg/mod/github.com/golang/glog@v0.0.0-20160126235308-23def4e6c14b/glog_file.go:63 +0x92 fp=0xc00002d860 sp=0xc00002d808 pc=0xe8da72
runtime.doInit(0x2f902e0)
        /usr/local/go/src/runtime/proc.go:6321 +0x126 fp=0xc00002d990 sp=0xc00002d860 pc=0x4d2246
runtime.doInit(0x2f92e60)
        /usr/local/go/src/runtime/proc.go:6298 +0x71 fp=0xc00002dac0 sp=0xc00002d990 pc=0x4d2191
runtime.doInit(0x2f8a980)
        /usr/local/go/src/runtime/proc.go:6298 +0x71 fp=0xc00002dbf0 sp=0xc00002dac0 pc=0x4d2191
runtime.doInit(0x2f8ec00)
        /usr/local/go/src/runtime/proc.go:6298 +0x71 fp=0xc00002dd20 sp=0xc00002dbf0 pc=0x4d2191
runtime.doInit(0x2f96a20)
        /usr/local/go/src/runtime/proc.go:6298 +0x71 fp=0xc00002de50 sp=0xc00002dd20 pc=0x4d2191
runtime.doInit(0x2f8bb80)
        /usr/local/go/src/runtime/proc.go:6298 +0x71 fp=0xc00002df80 sp=0xc00002de50 pc=0x4d2191
runtime.main()
        /usr/local/go/src/runtime/proc.go:233 +0x1d3 fp=0xc00002dfe0 sp=0xc00002df80 pc=0x4c4f53
runtime.goexit()
        /usr/local/go/src/runtime/asm_amd64.s:1594 +0x1 fp=0xc00002dfe8 sp=0xc00002dfe0 pc=0x4f6561

goroutine 2 [force gc (idle)]:
runtime.gopark(0x0?, 0x0?, 0x0?, 0x0?, 0x0?)
        /usr/local/go/src/runtime/proc.go:363 +0xd6 fp=0xc000064fb0 sp=0xc000064f90 pc=0x4c5356
runtime.goparkunlock(...)
        /usr/local/go/src/runtime/proc.go:369
runtime.forcegchelper()
        /usr/local/go/src/runtime/proc.go:302 +0xad fp=0xc000064fe0 sp=0xc000064fb0 pc=0x4c51ed
runtime.goexit()
        /usr/local/go/src/runtime/asm_amd64.s:1594 +0x1 fp=0xc000064fe8 sp=0xc000064fe0 pc=0x4f6561
created by runtime.init.6
        /usr/local/go/src/runtime/proc.go:290 +0x25

goroutine 3 [GC sweep wait]:
runtime.gopark(0x1?, 0x0?, 0x0?, 0x0?, 0x0?)
        /usr/local/go/src/runtime/proc.go:363 +0xd6 fp=0xc000065790 sp=0xc000065770 pc=0x4c5356
[... 97 lines cut ...]
GOSUMDB="off"
GOTMPDIR=""
GOTOOLDIR="/usr/local/go/pkg/tool/linux_amd64"
GOVCS=""
GOVERSION="go1.20.2"
GCCGO="gccgo"
GOAMD64="v1"
AR="ar"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
GOMOD="/dev/null"
GOWORK=""
CGO_CFLAGS="-O2 -g"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-O2 -g"
CGO_FFLAGS="-O2 -g"
CGO_LDFLAGS="-O2 -g"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m64 -pthread -Wl,--no-gc-sections -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build1815353546=/tmp/go-build -gno-record-gcc-switches"
```

I couldn't find a solution to this error. Many people seem to have it but not all on 'lookupUnixUid' function (as shown in the stacktrace). It looks like a go issue but I don't know how to continue from this point.

I am running this on LXC on proxmox. I don't want to use docker.

cedricve, 2023-04-07:
Thanks for reporting @AntonVonDelta  needs to be investigated!

cedricve, 2023-06-28:
The issue is resolved in latest build it was due to a missing link while building the static build.
https://github.com/kerberos-io/agent/commit/bf97bd72f18989a8dda51835c399db939b177996

closed 2023-06-28 by cedricve