kerberos-io/agent #94: Unable to update mounted config.json

b3ddy, 2023-04-02:
I'm having a similar issue to the currently open issue regarding pixel region not updating in the mounted config. However, with my mounted config none of the settings are saving from the configuration panel. Total noob here, was I supposed to give the container permissions for it to be allowed to update the config.json on the host machine?

cedricve, 2023-04-04:
Not sure if this is linked -> https://github.com/kerberos-io/agent/issues/80

b3ddy, 2023-04-06:
Thanks for the link, I'll test it asap and let you know.

b3ddy, 2023-04-12:
Hey @cedricve, 

Thanks for the link! My user owned the directory, but telling the container to use the host credentials seems to have fixed it. 

Ran the container with these settings and seems to be saving properly now. 

`docker run -p 80:80 --name mycamera \
-v /home/user/agent/config:/home/agent/data/config \
-v /home/user/agent/recordings:/home/agent/data/recordings \
--user 1000:1000 \
-d --restart=always kerberos/agent:latest
`

closed 2023-04-12 by b3ddy

instantdreams, 2023-04-12:
Makes sense. I'll try that if I spin up another kerberos agent.