kerberos-io/agent #98: Video looks like timelapse footage

b3ddy, 2023-04-25:
Hi, 

I'm running into an issue where the recordings look like timelapse footage. The record settings are set for 30sec, but when I go to video they are ~2-3 seconds long. The downloaded video appears to be even shorter at ~1sec. I've uploaded a quick video demonstrating the issue here https://vimeo.com/821003839?share=copy 

Screenshots of settings here: 
Recording - 
[image]

Conditions - 
[image]

Any help would be greatly appreciated! Thanks!

cedricve, 2023-04-25:
Interesting, this looks like frame loss (due to network connectivity). Can you share the logs generated by the agent? What kind of camera are you using, how is it connected?

b3ddy, 2023-04-26:
Hey @cedricve, 

Hope this helps, this is the log generated by the agent docker container. 

`         _  __         _                          _
        | |/ /___ _ __| |__   ___ _ __ ___  ___  (_) ___
        | ' // _ \ '__| '_ \ / _ \ '__/ _ \/ __| | |/ _ \
        | . \  __/ |  | |_) |  __/ | | (_) \__ \_| | (_) |
        |_|\_\___|_|  |_.__/ \___|_|  \___/|___(_)_|\___/

{"level":"info","msg":"Printing out environmentVariables (AGENT_...): ","time":"2023-04-18T20:39:18+02:00"}
{"level":"info","msg":"Successfully Opened config.json from opensource","time":"2023-04-18T20:39:18+02:00"}
{"level":"info","msg":"Printing our configuration (config.json): Type:  Key: LFTF6A75MM31X7ZBWYD3LLTOOAOXOG Name: >
{"level":"error","msg":"Checking data directory permissions: Problem creating a file: open ./data/snapshots/.test:>
[GIN-debug] [WARNING] Creating an Engine instance with the Logger and Recovery middleware already attached.

[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:   export GIN_MODE=release
 - using code:  gin.SetMode(gin.ReleaseMode)

[GIN-debug] GET    /debug/pprof/             --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/cmdline      --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/profile      --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] POST   /debug/pprof/symbol       --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/symbol       --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/trace        --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/allocs       --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/block        --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/goroutine    --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/heap         --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
{"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2023-04-18T14:39:18-04:00"}
[GIN-debug] GET    /debug/pprof/mutex        --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/threadcreate --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /swagger/*any             --> github.com/swaggo/gin-swagger.CustomWrapHandler.func1 (4 handlers)
[GIN-debug] GET    /ws                       --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] GET    /config                   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] POST   /config                   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] POST   /api/login                --> github.com/appleboy/gin-jwt/v2.(*GinJWTMiddleware).LoginHandler-f>
[GIN-debug] GET    /api/dashboard            --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] POST   /api/latest-events        --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] GET    /api/days                 --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] GET    /api/config               --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] POST   /api/config               --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] GET    /api/restart              --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] GET    /api/stop                 --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] POST   /api/hub/verify           --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] POST   /api/persistence/verify   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] GET    /api/stream               --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes>
[GIN-debug] POST   /api/camera/onvif/login   --> github.com/kerberos-io/agent/machinery/src/routers/http.LoginToOn>
[GIN-debug] POST   /api/camera/onvif/capabilities --> github.com/kerberos-io/agent/machinery/src/routers/http.GetO>
[GIN-debug] POST   /api/camera/onvif/pantilt --> github.com/kerberos-io/agent/machinery/src/routers/http.DoOnvifPa>
[GIN-debug] POST   /api/camera/onvif/zoom    --> github.com/kerberos-io/agent/machinery/src/routers/http.DoOnvifZo>
[GIN-debug] POST   /api/camera/verify/:streamType --> github.com/kerberos-io/agent/machinery/src/capture.VerifyCam>
[GIN-debug] GET    /file/*filepath           --> github.com/kerberos-io/agent/machinery/src/routers/http.Files (9 >
[GIN-debug] [WARNING] You trusted all proxies, this is NOT safe. We recommend you to set a value.
Please check https://pkg.go.dev/github.com/gin-gonic/gin#readme-don-t-trust-all-proxies for details.
[GIN-debug] Listening and serving HTTP on :80
[{video H264 0 90000 rtsp://192.168.0.21:8554/cam/trackID=0 96 0 [] [[39 77 64 41 149 160 30 0 137 249 97 0 0 3 0 >
{"level":"info","msg":"RunAgent: opened RTSP streamrtsp://192.168.0.21:8554/cam","time":"2023-04-18T14:39:18-04:00>
{"level":"info","msg":"RunAgent: camera settings changed, reloading decoder","time":"2023-04-18T14:39:18-04:00"}
[... cut ...]
{"level":"info","msg":"ControlAgent: Number of packets read 2","time":"2023-04-18T14:39:23-04:00"}
{"level":"info","msg":"HandleStream: packet size 133931","time":"2023-04-18T14:39:24-04:00"}
{"level":"info","msg":"HandleStream: packet size 133433","time":"2023-04-18T14:39:26-04:00"}
{"level":"info","msg":"HandleStream: packet size 133108","time":"2023-04-18T14:39:28-04:00"}
{"level":"info","msg":"ControlAgent: Number of packets read 5","time":"2023-04-18T14:39:28-04:00"}
[GIN] 2023/04/18 - 18:39:29 | 304 |     432.926µs |    192.168.0.14 | GET      "/"
`

[image]

cedricve, 2023-05-10:
@b3ddy, does the Agent and RPI camera run on the same board? You are using the sidecar container to proxy to an RTSP stream?

b3ddy, 2023-05-10:
Hey @cedricve Yes, both are running on Raspberry Pi 4 with RPICamv3. Utilizing aler9/rtsp-simple-server:latest-rpi(Image ID bf58f0231d60) docker image as the sidecar to provide the agent with RTSP stream url and the agent docker image is kerberos/agent:latest(Image ID bd75dfb8182a). Knowing that the camera and agent are on the same board, network connectivity shouldn't be an issue, right?

cedricve, 2023-05-10:
Exactly! I do run the same setup on one of my boards. Only difference is that we I this image: kerberos/rtsp-simple-server:latest-rpi.

Can you give that one a try?