kerberos-io/agent #99: hangs both firefox and safari

telefax, 2023-04-26:
I tried this using a docker container.

Just visiting the dashboard page manage to hang both safari and firefox, thats...uhm..impressive. Impossible to do anything else.

cedricve, 2023-04-27:
Wouw that's impressive 🤩. Which version are you using? Latest?

What happens if you visit our demo.kerberos.io page? Cannot replicate.

Frankkkkk, 2023-05-30:
Hi, I somehow have the same problem. I was trying the agent and it makes the 3 major browsers grind to a halt (safari, chrome, ff). Here is a small performance screenshot:
[image]
My best bet is because I misconfigured the agent, as show in the logs:
```

```

cedricve, 2023-05-30:
Thanks for the detailed report! Does it recover after a page refresh or it stays halted? 

> {"level":"error","msg":"Something went wrong while opening RTSP: rtsp: no username","

Frankkkkk, 2023-05-30:
In my case it always grinds the browser to a halt because my reolink camera doesn't seem compatible. It only happens on /dashboard though.

I'll try with another asap.

cedricve, 2023-05-30:
I'll look into this! So the camera is blocking the websocket, let's see what it does :)

cedricve, 2023-05-30:
Thanks, will have a look. @Frankkkkk which Reolink are you using?

cedricve, 2023-05-30:
Hmm cannot replicate with the latest `kerberos:agent` docker image @Frankkkkk. If anyone has instructions how to simulate would help.

Frankkkkk, 2023-05-30:
I'm also using `docker.io/kerberos/agent@sha256:6ff7365dbc784613671d9afd8c2038dbc11506ead659ea054e6c8ed9ecd82e4c`.  Here are some env values that cause the problem for me:
```yaml
        - name: AGENT_NAME                                                         
          value: demo-agent                                                        
        - name: AGENT_CAPTURE_IPCAMERA_RTSP                                        
          value: fixme                                                             
        - name: AGENT_HUB_KEY                                                      
          value: yyy                                                               
        - name: AGENT_HUB_PRIVATE_KEY                                              
          value: yyy                                                               
```
[image]

And the full container logs:

<details>
  <summary>Click me</summary>
  

```
	 _  __         _                          _       
	| |/ /___ _ __| |__   ___ _ __ ___  ___  (_) ___  
	| ' // _ \ '__| '_ \ / _ \ '__/ _ \/ __| | |/ _ \ 
	| . \  __/ |  | |_) |  __/ | | (_) \__ \_| | (_) |
	|_|\_\___|_|  |_.__/ \___|_|  \___/|___(_)_|\___/ 
													  
	
{"level":"info","msg":"Printing out environmentVariables (AGENT_...): AGENT_CAPTURE_IPCAMERA_RTSP=fixme AGENT_HUB_KEY=yyy AGENT_HUB_PRIVATE_KEY=yyy AGENT_NAME=demo-agent AGENT_SVC_SERVICE_PORT=80 AGENT_SVC_PORT=tcp://10.43.88.99:80 AGENT_SVC_PORT_80_TCP_PORT=80 AGENT_SVC_PORT_80_TCP_ADDR=10.43.88.99 AGENT_SVC_PORT_80_TCP_PROTO=tcp AGENT_SVC_SERVICE_HOST=10.43.88.99 AGENT_SVC_PORT_80_TCP=tcp://10.43.88.99:80 ","time":"2023-05-30T21:02:13+02:00"}
{"level":"info","msg":"Successfully Opened config.json from opensource","time":"2023-05-30T21:02:13+02:00"}
{"level":"info","msg":"Printing our configuration (config.json): Type:  Key:  Name: demo-agent FriendlyName:  Time: false Offline: false AutoClean: true RemoveAfterUpload: true MaxDirectorySize: 100 Timezone: Africa/Ceuta Capture: { {fixme   false   } {} {} true true true true false 20 10 30  0  false 8 150} Timetable: [0xc000414480 0xc0004144a0 0xc0004144c0 0xc0004144e0 0xc000414500 0xc000414520 0xc000414540] Region: \u0026{ {0 0 800 640} []} Cloud: s3 S3: \u0026{ http://proxy.kerberos.io kerberosaccept eu-west1   } KStorage: \u0026{     } Dropbox: \u0026{ } MQTTURI: tcp://mqtt.kerberos.io:1883 MQTTUsername:  MQTTPassword:  STUNURI: stun:turn.kerberos.io:8443 TURNURI: turn:turn.kerberos.io:8443 TURNUsername: username1 TURNPassword: password1 HeartbeatURI:  HubURI: https://api.cloud.kerberos.io HubKey: yyy HubPrivateKey: yyy HubSite:  ConditionURI:  ","time":"2023-05-30T21:02:13+02:00"}
{"level":"info","msg":"Checking data directory permissions: OK","time":"2023-05-30T21:02:13+02:00"}
{"level":"info","msg":"Main: updated unique key for agent to: 67SZ1HIVSF7U1BK7FQGUWQ00QS9O3W","time":"2023-05-30T21:02:13+02:00"}
[GIN-debug] [WARNING] Creating an Engine instance with the Logger and Recovery middleware already attached.

[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:	export GIN_MODE=release
 - using code:	gin.SetMode(gin.ReleaseMode)

[GIN-debug] GET    /debug/pprof/             --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/cmdline      --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
{"level":"error","msg":"Something went wrong while opening RTSP: open fixme: no such file or directory","time":"2023-05-30T21:02:13+02:00"}
{"level":"info","msg":"ControlAgent: Number of packets read 0","time":"2023-05-30T21:02:13+02:00"}
[GIN-debug] GET    /debug/pprof/profile      --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] POST   /debug/pprof/symbol       --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/symbol       --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/trace        --> github.com/gin-gonic/gin.WrapF.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/allocs       --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/block        --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/goroutine    --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/heap         --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/mutex        --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /debug/pprof/threadcreate --> github.com/gin-gonic/gin.WrapH.func1 (3 handlers)
[GIN-debug] GET    /swagger/*any             --> github.com/swaggo/gin-swagger.CustomWrapHandler.func1 (4 handlers)
[GIN-debug] GET    /ws                       --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func1 (4 handlers)
[GIN-debug] GET    /config                   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func2 (4 handlers)
[GIN-debug] POST   /config                   --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func3 (4 handlers)
[GIN-debug] POST   /api/login                --> github.com/appleboy/gin-jwt/v2.(*GinJWTMiddleware).LoginHandler-fm (4 handlers)
[GIN-debug] GET    /api/dashboard            --> github.com/kerberos-io/agent/machinery/src/routers/http.AddRoutes.func4 (4 handlers)
[... cut ...]
{"level":"error","msg":"Something went wrong while opening RTSP: open fixme: no such file or directory","time":"2023-05-30T21:03:07+02:00"}
{"level":"info","msg":"Successfully Opened config.json from opensource","time":"2023-05-30T21:03:10+02:00"}
{"level":"error","msg":"Something went wrong while opening RTSP: open fixme: no such file or directory","time":"2023-05-30T21:03:10+02:00"}
{"level":"info","msg":"Successfully Opened config.json from opensource","time":"2023-05-30T21:03:13+02:00"}
```
</details>

cedricve, 2023-06-08:
We added a fix which unblocks when using an unsupported codec (h265). Would you be able to verify if the web page still hangs with your previous setup using the `latest` docker image? Thank you so much @Frankkkkk

closed 2023-06-28 by cedricve