kerberos-io/agent #105: SDP malformed: sprop-parameter-sets=

0xhanh, 2023-06-28:
Agent raise error: "Error: rtsp: missing h264 sps or pps" when deal with H264 tracks that have an empty sprop-parameter-sets
same as link: https://github.com/bluenviron/mediamtx/issues/861

Anyone tried to fix it?

cedricve, 2023-06-28:
@0xhanh RTSP server is required to set both SPS and PPS right? I might miss some context here.

0xhanh, 2023-06-28:
@cedricve:  Both SPS and PPS, it's quite common to deal with H264 tracks that have an empty sprop-parameter-sets, for various reasons - it may be caused be a defective software/firmware or by choice since parameters may change during time. Some cameras don't provide it. 

These (SPS, PPS) can be discovered from H264 tracks? gortsplib fixed this [issue](https://github.com/bluenviron/mediamtx/commit/3e8668f9e242256f641bb6be6d2421cc4b97432c).

Furthermore, cameras can do change SPS and PPS without closing the stream, so that:

* cameras that insert SPS and PPS into the SDP and do not send SPS and PPS NALUs
* cameras that insert SPS and PPS into the SDP, send SPS and PPS NALUs and can change them
* cameras that don't insert SPS and PPS into the SDP, send SPS and PPS NALUs and can change them

Therefore, the Agent will be adapter to support all these scenarios and will edit the SDP according to the last received SPS and PPS 
(From prob: https://github.com/bluenviron/mediamtx/issues/787)

cedricve, 2023-07-10:
thanks for the detailed explanation @0xhanh! Will check with the time if we will make the adaption in the joy4 library or the new media library we are working on!

cedricve, 2023-10-31:
Looks like this issue is occuring with Hanwha cameras.

cedricve, 2023-10-31:
We'll prioritise this @0xhanh, can you verify which camera raised this error?

cedricve, 2023-11-01:
Looks like you have fixed it already in your fork! Wonderful work, can you confirm this resolved your issue with the missing SSP, PPS: https://github.com/0xhanh/joy4/commit/68dbc5fd2cb3de57844844a1e21236fa2f544081

klihub, 2023-11-18:
> We'll prioritise this @0xhanh, can you verify which camera raised this error?

@cedricve The exact same error happens also with 'Foscam  D2EP FHD IP Camera'.

cedricve, 2024-01-08:
We embedded a new rtsp library, might have fixed this issue. Closing for now, please reopen if still the case.