kerberos-io/agent #109: Kerberos Agent not retaining setting across reboots.

kprose3154, 2023-07-12:
Hi everyone,

So I just started working on getting Kerberos working at home and I've ran into a big issue. It's possible I misconfigured something but from how to install guide goes, I did not miss anything. When I deploy a camera agent using Factory it works, until I reboot the system. After everything starts back up none of the settings for the agent are there anymore. I'm still able to see it in Factory just all of the camera settings are missing. Can anyone help with this?

cedricve, 2023-07-12:
@kprose3154  can you confirm the mongodb pod is still working?

kprose3154, 2023-07-13:
@cedricve I can confirm it is still working. I'm able to port-forward the pod and talk to it using compass. The only collection in the KeberosFactory schema is configuration. What I'm noticing though is the information gets pushed like normal and it shows up in the DB but as soon as I reboot all that data goes away. The document is still present but all the variables are blank. (Second Screenshot)
<img width="1458" alt="Screenshot 202
[image]
3-07-12 at 10 31 53 PM" src="https://github.com/kerberos-io/agent/assets/33162479/ac957f61-7fb5-49ca-b430-bd6ad90a8bb6">

cedricve, 2023-07-13:
This looks like a MongoDB/persistence issue , how are you persisting your data, are you using OpenEBS as well?

kprose3154, 2023-07-13:
I am using OpenEBS, at the default path /var/openebs/local/. It is only the cameras getting lost, the global config stays around after reboots.
[image]

cedricve, 2023-07-14:
Thank you, the thing is that the configuration collection is auto created on start when the collection is not available.

Can you create a collection in your mongodb manually and insert some data into it? the  restart the machine and check if it still persist. I strongly believe the underlaying PV is not persisting properly.

kprose3154, 2023-07-17:
I created both a temp DB (Screenshot 2) and a temp collection in the KerberosFactory DB (Screenshot 1) and all the data persisted. Maybe I have something misconfigured but I followed the guide. The only thing I changed was the namespace things were put under so instead of KerberosFactory I put them under Kerberos. MongoDB is still in its own namespace as helm puts it there.
[image]
[image]

cedricve, 2023-07-17:
Thank you for validating, this is interesting. So once you add a new Kerberos Agent through the UI it will boot up, but it will not persist it configuration in the mongodb.

Are you able to modify the "global" settings page of the Kerberos Factory and reflect your changes in the configuration setting?

kprose3154, 2023-07-18:
Yes, the agent pod will always boot up. The only difference is the configuration for it no longer exists in the DB after reboots, the config gets stored like it is supposed to before reboots. I am able to modify "global" settings and it does reflect in the DB (Screenshot 1). I bought a temp hub subscription to get keys to put in, and they persisted. So it only looks like the configuration for cameras gets erased, the global configuration persists.

[image]

kprose3154, 2023-07-28:
@cedricve Just checking to see if there is any update on this?

cedricve, 2023-07-30:
Will have a look @kprose3154, just came back from a long holiday! ;)

cedricve, 2023-09-07:
@kprose3154 is your system/behaviour still the same?