kerberos-io/agent #111: No  variables PUID and PGID

neutralvibes, 2023-07-17:
Seems as if agents are forced to run as 'root' in container. Not nice. Would be nicer if what now appears to be a standard is used.

'''yam

environment:
  PUID=1005 # User
  PGID=102 # Group
  othervars=....
'''

This will also allow for required user access outside the container without requiring root. Forgive me if there is already a way to do this.

yllekz, 2023-08-19:
Yes please add PUID and GID variables. I recently moved my agent docker container to openmediavault but I think it's failing to record events due to permissions problems and normally this is fixed by having the user/group ID configurable via variables (see also linuxserver's docker containers)

cedricve, 2023-08-20:
Would this be sufficient (https://docs.linuxserver.io/general/understanding-puid-and-pgid). It mentions:

    We are aware that recent versions of the Docker engine have introduced the --user flag. Our images are not yet compatible with this, so we recommend continuing usage of PUID and PGID.

Would this work as well? https://docs.docker.com/engine/reference/run/#user

cedricve, 2023-09-07:
@yllekz  @neutralvibes  any thoughts?

winston0410, 2024-06-28:
Just came across this issue, for K8S if I set `runAsNonRoot: true`, I will also need to set `runAsUser: <uid>`.  @cedricve From the dockerfile I can see there is an non-root user, but there seems not to be an uid set for the user? A static UID would be nice and easier for config