kerberos-io/agent #114: Agents not shown in Hub

KickWilstra, 2023-09-05:
After using kerberos.io for several years i had to install kerberos.io all over again because of broken hardware. I installed two agents on a raspberry pi4b using raspbarian os (32bit) and docker compose. Both agents are running fine and next i tried to connect the agents to the hub (app.kerberos.io). I had an old subscription plan so i choose a new plan (pro). The pro plan is btw shown as gold in the subscription overview which is confusing. Accordingly i copied the private and public keys in the agents and verified the connections. As wel the connection to the hub as the connection to the storage where fine and so i stored the settings. But when i look into the hub i don't see the agents appear. 
In the logging of the rapsberry pi i see on a regular base the following error message.
 {"level":"error","msg":"HandleHeartBeat: (400) Something went wrong while sending to Kerberos Hub.","time":"2023-09-05T16:03:54+01:00"}

Any idea what can cause this problem?

KickWilstra, 2023-09-06:
Here's an update about the problem i'm experiencing. This morning i noticed i've become new keys for the hub connection. After applying them the recordings are send to the hub and i'm able to play them in the hub. 
The only thing that's missing is that my camera's don't show up at the connected camera's item.  So liveview doesn't work either. The error message mentioned in my previous message is also still there.

cedricve, 2023-09-07:
Correct, we've rolled out a fix for this. Something was broken due to integration of ONVIF presets. When a camera has no PTZ configuration or ONVIF, the Kerberos Hub connection was failing. This is now fixed in https://github.com/kerberos-io/agent/commit/6325e37aaed8f524ab99c7307f3c625ef9fefcba.

KickWilstra, 2023-09-12:
Thanks for your fast response Cedric. Agents are connected to the hub now. Unfortunately another issue did show up. When i select liveview in the hub the 'running circle' shows up but the liveview window stays black and the circle is running forever. Selecting SD or HD doesn't make a difference though i configured both streams.

cedricve, 2023-09-12:
> You are receiving this email because you commented on the thread.

Can you send over the logs? Because that’s not expected. Do you see the Agent is properly running, you should see a series of packet received logs.

KickWilstra, 2023-09-12:
I've two cameras running and in one of the windows i have the following message "The agent is online, but the camera stream seems to be offline"
Agents are properly running and the liveview in the agents are fine, Also sending the images to the hub are working fine.

Is there an easy way to get the log files or do i have to look into the containers? I'm not a linux guru so have to find out how to do that.

cedricve, 2023-09-12:
It looks like 1 of those agents is not working as expected (faulty stream? is it h264?). You can find the logs by `docker logs <container_name/container_id>`. Where you can find the name or if of the container with `docker ps -a`

KickWilstra, 2023-09-12:
I found the log and remarkable is the following message
{"level":"info","msg":"UploadKerberosHub: Upload NOT allowed using the credentials provided (public key, private keyJ)","time":"2023-09-11T19:14:32+02:00"}
I changed the keys in this message off course. I you need the complete logfile i can mail it to you (what's address?).

cedricve, 2023-09-12:
Thank you @KickWilstra  you can send it to `support@kerberos.io`.

KickWilstra, 2023-09-12:
Has been sent

KickWilstra, 2023-09-26:
@cedricve live video on the hub is still not working. Any suggestions what can be wrong?