kerberos-io/agent #117: GetPTZConfigurationsFromDevice:  error in NodeName - username and password might be wrong

128keaton, 2023-09-26:
Hello,  I am using a Hikvision DS-2CD2643G2-IZS. With ONVIF enabled, I see the following error in my agents' logs:
```
kerberos-agents-kerberos-agent1-1  | {"level":"error","msg":"GetPTZConfigurationsFromDevice: error in NodeName - username and password might be wrong","time":"2023-09-26T23:04:02+02:00"}
kerberos-agents-kerberos-agent1-1  | {"level":"error","msg":"HandleHeartBeat: error while getting PTZ configurations: error in NodeName - username and password might be wrong","time":"2023-09-26T23:04:02+02:00"}
```

I'm not able to discern where the issue is, however the following is the device capabilities response
<details><summary>Device Capabilities Response</summary>
<p>

```xml
<?xml version="1.0" encoding="UTF-8"?>
<env:Envelope xmlns:env="http://www.w3.org/2003/05/soap-envelope" xmlns:soapenc="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:tan="http://www.onvif.org/ver20/analytics/wsdl" xmlns:tst="http://www.onvif.org/ver10/storage/wsdl" xmlns:ter="http://www.onvif.org/ver10/error" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:tmd="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:wsdl="http://schemas.xmlsoap.org/wsdl" xmlns:wsoap12="http://schemas.xmlsoap.org/wsdl/soap12" xmlns:http="http://schemas.xmlsoap.org/wsdl/http" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:wsadis="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsa="http://www.w3.org/2005/08/addressing" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:wsrf-bf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wsntw="http://docs.oasis-open.org/wsn/bw-2" xmlns:wsrf-rw="http://docs.oasis-open.org/wsrf/rw-2" xmlns:wsaw="http://www.w3.org/2006/05/addressing/wsdl" xmlns:wsrf-r="http://docs.oasis-open.org/wsrf/r-2" xmlns:trc="http://www.onvif.org/ver10/recording/wsdl" xmlns:tse="http://www.onvif.org/ver10/search/wsdl" xmlns:trp="http://www.onvif.org/ver10/replay/wsdl" xmlns:tnshik="http://www.hikvision.com/2011/event/topics" xmlns:hikwsd="http://www.onvifext.com/onvif/ext/ver10/wsdl" xmlns:hikxsd="http://www.onvifext.com/onvif/ext/ver10/schema" xmlns:tas="http://www.onvif.org/ver10/advancedsecurity/wsdl" xmlns:tr2="http://www.onvif.org/ver20/media/wsdl" xmlns:axt="http://www.onvif.org/ver20/analytics">
  <env:Body>
    <tds:GetCapabilitiesResponse>
      <tds:Capabilities>
        <tt:Analytics>
          <tt:XAddr>http://10.1.20.80/onvif/Analytics</tt:XAddr>
          <tt:RuleSupport>true</tt:RuleSupport>
          <tt:AnalyticsModuleSupport>true</tt:AnalyticsModuleSupport>
        </tt:Analytics>
        <tt:Device>
          <tt:XAddr>http://10.1.20.80/onvif/device_service</tt:XAddr>
          <tt:Network>
            <tt:IPFilter>true</tt:IPFilter>
            <tt:ZeroConfiguration>true</tt:ZeroConfiguration>
            <tt:IPVersion6>true</tt:IPVersion6>
            <tt:DynDNS>true</tt:DynDNS>
            <tt:Extension>
              <tt:Dot11Configuration>false</tt:Dot11Configuration>
              <tt:Extension>
                <tt:DHCPv6>true</tt:DHCPv6>
                <tt:Dot1XConfigurations>0</tt:Dot1XConfigurations>
              </tt:Extension>
            </tt:Extension>
          </tt:Network>
          <tt:System>
            <tt:DiscoveryResolve>false</tt:DiscoveryResolve>
            <tt:DiscoveryBye>true</tt:DiscoveryBye>
            <tt:RemoteDiscovery>false</tt:RemoteDiscovery>
            <tt:SystemBackup>false</tt:SystemBackup>
            <tt:SystemLogging>true</tt:SystemLogging>
            <tt:FirmwareUpgrade>true</tt:FirmwareUpgrade>
            <tt:SupportedVersions>
              <tt:Major>19</tt:Major>
              <tt:Minor>12</tt:Minor>
            </tt:SupportedVersions>
            <tt:SupportedVersions>
              <tt:Major>16</tt:Major>
              <tt:Minor>12</tt:Minor>
            </tt:SupportedVersions>
            <tt:SupportedVersions>
              <tt:Major>2</tt:Major>
              <tt:Minor>60</tt:Minor>
            </tt:SupportedVersions>
            <tt:SupportedVersions>
              <tt:Major>2</tt:Major>
              <tt:Minor>40</tt:Minor>
            </tt:SupportedVersions>
            <tt:SupportedVersions>
              <tt:Major>2</tt:Major>
              <tt:Minor>20</tt:Minor>
            </tt:SupportedVersions>
            <tt:SupportedVersions>
              <tt:Major>2</tt:Major>
              <tt:Minor>10</tt:Minor>
            </tt:SupportedVersions>
            <tt:SupportedVersions>
              <tt:Major>2</tt:Major>
              <tt:Minor>0</tt:Minor>
[... 89 lines cut ...]
          </tt:Search>
          <tt:Replay>
            <tt:XAddr>http://10.1.20.80/onvif/Replay</tt:XAddr>
          </tt:Replay>
        </tt:Extension>
      </tds:Capabilities>
    </tds:GetCapabilitiesResponse>
  </env:Body>
</env:Envelope>
```

</p>
</details> 

And lastly the response from requesting PTZ Service Capabilities:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<env:Envelope xmlns:env="http://www.w3.org/2003/05/soap-envelope" xmlns:soapenc="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:tan="http://www.onvif.org/ver20/analytics/wsdl" xmlns:tst="http://www.onvif.org/ver10/storage/wsdl" xmlns:ter="http://www.onvif.org/ver10/error" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:tmd="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:wsdl="http://schemas.xmlsoap.org/wsdl" xmlns:wsoap12="http://schemas.xmlsoap.org/wsdl/soap12" xmlns:http="http://schemas.xmlsoap.org/wsdl/http" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:wsadis="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsa="http://www.w3.org/2005/08/addressing" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:wsrf-bf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wsntw="http://docs.oasis-open.org/wsn/bw-2" xmlns:wsrf-rw="http://docs.oasis-open.org/wsrf/rw-2" xmlns:wsaw="http://www.w3.org/2006/05/addressing/wsdl" xmlns:wsrf-r="http://docs.oasis-open.org/wsrf/r-2" xmlns:trc="http://www.onvif.org/ver10/recording/wsdl" xmlns:tse="http://www.onvif.org/ver10/search/wsdl" xmlns:trp="http://www.onvif.org/ver10/replay/wsdl" xmlns:tnshik="http://www.hikvision.com/2011/event/topics" xmlns:hikwsd="http://www.onvifext.com/onvif/ext/ver10/wsdl" xmlns:hikxsd="http://www.onvifext.com/onvif/ext/ver10/schema" xmlns:tas="http://www.onvif.org/ver10/advancedsecurity/wsdl" xmlns:tr2="http://www.onvif.org/ver20/media/wsdl" xmlns:axt="http://www.onvif.org/ver20/analytics">
  <env:Body>
    <tptz:GetServiceCapabilitiesResponse>
      <tptz:Capabilities EFlip="false" Reverse="false" GetCompatibleConfigurations="true" MoveStatus="true" StatusPosition="false"/>
    </tptz:GetServiceCapabilitiesResponse>
  </env:Body>
</env:Envelope>
```

cedricve, 2023-09-29:
Does the Hikvision has PTZ enabled? I don't believe so, you might ignore the error than.

cedricve, 2024-01-03:
We have turned this into a warning as it doesnt affect the behaviour of the agent. Latest release will hide the error. Closing this issue for now.

closed 2024-01-03 by cedricve