kerberos-io/agent #120: Cannot watch recordings on phone or mac

VC-Semih, 2023-10-14:
Using https://github.com/kerberos-io/camera-to-rtsp with a Logitech C270
I can watch my recordings only on Windows (and probably Linux did not test) but not on Mac or Android phone I can only watch then if I download vlc and open the file with it.
Does someone have a clue about this ? Thanks !

PS: demo.kerberos.io works on mac so it could be due to camera-to-rstp ?

VC-Semih, 2023-10-14:
Here is the command line in the yml config:
ffmpeg -f v4l2 -i /dev/video0 -preset ultrafast -c:v libx264 -x264-params keyint=60:scenecut=0 -f rtsp rtsp://myip$RTSP_PORT/$RTSP_PATH

cedricve, 2023-10-16:
Hey @VC-Semih, interesting, what happens when you drag and drop these recordings into your browser. Do they also play then? 

- First download locally to your host system
- Open your browser, and drag you recording into the browser

VC-Semih, 2023-10-17:
Just tested downloading the file on windows, the "Movies & TV" App cannot read it too, I only have a black screen with the player stuck (video timecode not changing) but the browser can read it (with drag and drop as you suggested)
I don't know if the file will be processed by github but here is a video upload of a recording
https://github.com/kerberos-io/agent/assets/55637588/e1ff67ad-61fe-481c-b054-f9c35a23d901

cedricve, 2023-10-17:
Hmm I can play it from the browser 
[image]

VC-Semih, 2023-10-18:
Hmm that's strange, here is a screen capture on a Google pixel 6 with android 14: 

https://github.com/kerberos-io/agent/assets/55637588/362f547a-c6e5-4294-a727-4b033100fd85

VC-Semih, 2023-10-18:
For debugging purpose here is ffprobe result on a recording

```
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'video.mp4':
  Metadata:
    creation_time   : 2020-01-10T19:37:32.000000Z
  Duration: 00:00:40.00, start: 0.000000, bitrate: 1587 kb/s
  Stream #0:0[0x1](und): Video: h264 (High 4:2:2) (avc1 / 0x31637661), yuv422p(progressive), 640x480, 1585 kb/s, 29.98 fps, 30 tbr, 10000k tbn (default)
    Metadata:
      creation_time   : 2020-01-10T19:37:32.000000Z
      handler_name    : Handler
      vendor_id       : [0][0][0][0]
```

VC-Semih, 2023-10-19:
Little update, I did find a way to make recordings readable by Android and MacOs using `ffmpeg -f v4l2 -i /dev/video0 -preset ultrafast -c:v libx264 -profile:v high10 -vf format=yuv420p -f rtsp rtsp:192...` but the live video and the movement detection is no longer working 
[image]

VC-Semih, 2023-10-19:
After a lot of tinkering I have an ffmpeg command that works on Mac Android & Windows and works with Kerberos live stream and movement detection

`ffmpeg -f v4l2 -i /dev/video0 -preset ultrafast -c:v libx264 -x264-params keyint=60:scenecut=0 -vf format=yuv420p -f rtsp rtsp://192...`

closed 2023-10-19 by VC-Semih

yllekz, 2023-12-02:
@VC-Semih what environment variable does that ffmpeg command map to? I am having difficulties finding where this goes in the yml.

VC-Semih, 2023-12-06:
@yllekz using [camera-to-rtsp repo](https://github.com/kerberos-io/camera-to-rtsp), replace the "runOnInit: ffmpeg..." with the command above if you have similar problems to mine