kerberos-io/agent #126: Raspberry pi - Docker / ONVIF

unichargepay, 2023-11-09:
Hello.
I'm having issues with connecting to my IP Camera using ONVIF protocol.

**System info:**

''' 
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
'''

'''
docker version
Client: Docker Engine - Community
 Version:           24.0.6
 API version:       1.43
 Go version:        go1.20.7
 Git commit:        ed223bc
 Built:             Mon Sep  4 12:31:36 2023
 OS/Arch:           linux/arm64
 Context:           default

Server: Docker Engine - Community
 Engine:
  Version:          24.0.6
  API version:      1.43 (minimum version 1.12)
  Go version:       go1.20.7
  Git commit:       1a79695
  Built:            Mon Sep  4 12:31:36 2023
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.6.22
  GitCommit:        8165feabfdfe38c65b599c4993d227328c231fca
 runc:
  Version:          1.1.8
  GitCommit:        v1.1.8-0-g82f18fe
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
'''

Pretty much used the tutorial for seting up with docker-compose:

'''
version: "3.9"
services:
  kerberos-agent1:
    image: "kerberos/agent:latest"
    ports:
      - "8081:80"
    environment:
      - AGENT_NAME=agent1
      - AGENT_CAPTURE_IPCAMERA_ONVIF_XADDR=x.x.x.x:80
      - AGENT_CAPTURE_IPCAMERA_ONVIF_USERNAME=admin
      - AGENT_CAPTURE_IPCAMERA_ONVIF_PASSWORD=megapassword
    volumes:
      - ./agent1/config:/home/agent/data/config
      - ./agent1/recordings:/home/agent/data/recordings
'''

The configuration file I'm using is the same as in the tutorial.
[](https://raw.githubusercontent.com/kerberos-io/agent/master/machinery/data/config/config.json)

However, after runing compose up, there is an issue with the ONVIF endpoint:
'''
{"level":"error","msg":"HandleHeartBeat: error while getting PTZ configurations: target endpoint service not found","time":"2023-11-09T19:49:49+01:00"}
'''

Host is in the same network as the IP Camera.
I can ping the camera from the host.

Also I tried to connect to the camera with a ODM tool (ONVIF Device Manager) with the proper credentials and it works:

[image]

I have no idea, why this doesn't work.
Also I noticed, that any change I do, won't get saved. Not sure, if related.

Can, someone, please help me with this?

cedricve, 2023-11-10:
Hey there, the error you're saying is that PTZ is not available on the device (it's actually a warning). Can you confirm ONVIF, by pressing the "verify" button?

[image]

unichargepay, 2023-11-10:
Hello.
First of all, thank you for the support.

I managed to get around the permissions.
After changing some settings and compose down, the setting got stored in the config.json.

```
$ chmod -R 755 agent1
$ chown 100:101 agent1/ -R
```

With the ONVIF, i tried again, but still got the error.
[image]

The ONVIF is active on the IPCAM, as I posted above, i can connect to the camera with different app.
Is the ONVIF used only for the PTZ?

I'm trying to build an application, that will grab a list of IP addresses from a database and spin up agents for each camera
that were preconfigured with credentials for the ONVIF.
Can this be done, or do I have to know specific RTSP URLs?

My NAS from synology can get a streams and other function configured automaticaly just through ONVIF.
Sorry, i'm trying to get my head around this, but my only chance to get this up and running on the scale is using Kerberos nad it's API for the function that I will need in the future.

If I can get this running, i will get around figuring out the licensing.

Thanks, again.

cedricve, 2023-11-10:
Currently we use ONVIF for PTZ through our [Kerberos Hub](https://github.com/kerberos-io/hub) offering. It looks like the ONVIF library in the Kerberos Agent is not able to detect the ONVIF settings of the camera you are using (would need to debug it).  Currently we don't leverage RTSP discovery in the agent through ONVIF (we do it in [Kerberos Factory](https://github.com/kerberos-io/factory)), so for now you'll need to fetch the RTSP stream manually when using our Kerberos Agent.

Regarding your use case, you could build ONVIF around it. In your application you run ONVIF discovery (through another library), fetch the IP address and RTSP stream of each camera, store it in your database, and then start a Kerberos Agent ([using environment variables](https://github.com/kerberos-io/agent#configure-with-environment-variables))

closed 2023-12-03 by unichargepay