kerberos-io/agent #127: Not an issue, but a question

deanfourie1, 2023-11-23:
Sorry not really sure where else to put this but,

I am trying to choose the correct version. 
My setup is around 10 cams and for onsite recording only with no internet access.

Will Agent work for this?

Thanks

cedricve, 2023-12-03:
hey @deanfourie1,

> My setup is around 10 cams and for onsite recording only with no internet access.

I would go for a [docker compose setup](https://github.com/kerberos-io/agent/tree/master/deployments/docker#2-running-multiple-containers-with-docker-compose). 

> Also, is there a bare metal ISO I can flash rather then going the docker route?

We try to avoid bare metal ISO or VMs in favour of Docker. What is your requirement?

deanfourie1, 2023-12-07:
Yea I've gone the docker route but just struggling with multiple disks.

Is there a way to mount and pass through multiple disks for additional storage?

Thanks

cedricve, 2024-01-03:
you might want to setup Minio object storage, which would consolidate your disks and make them available through a "single disk" in Kerberos Vault.