kerberos-io/agent #129: Config file is not found ./data/config/config.json

chenxudong2020, 2023-11-29:
After executing the container startup using the following command, the log error is as follows

[image]

docker run -p 9999:80 --name mycamera \
-e AGENT_USERNAME=rr\
-e AGENT_PASSWORD=rr@@ \
-e AGENT_TIMEZONE=GMT+8 \
-v $(pwd)/agent/config:/home/agent/data/config \
-v $(pwd)/agent/recordings:/home/agent/data/recordings \
-d --restart=always kerberos/agent:latest

cedricve, 2023-12-03:
hey @chenxudong2020  did you copy the `config.json`?

chenxudong2020, 2023-12-04:
I couldn't find this file, so I didn't copy it either. Your answer has already been found. Thank you

chenxudong2020, 2023-12-04:
According to your steps, you added config.json and restarted the container with an error message

[image]

cedricve, 2023-12-29:
Looks like the timezone isn't set correctly.

HugoDL, 2024-06-20:
I had the same problem running on Docker. To fix it, I created the config.json file using this one as an example: [config.json](https://github.com/kerberos-io/agent/blob/master/machinery/data/config/config.json)