kerberos-io/agent #130: Snapshot

unichargepay, 2023-12-03:
Hello.

Is it possible to trigger a capture of a snapshot by API, or through MQTT?
If so, I coulnd't find any kind of documentation.

Thank you.

unichargepay, 2023-12-21:
Hello.
I was wondering, if this is possible with use of additional packages by Kerberos.io?

cedricve, 2023-12-21:
It will be added in our new version (published end of week), we will have a API endpoint to request a snapshot. Hope this helps.

unichargepay, 2023-12-24:
> It will be added in our new version (published end of week), we will have a API endpoint to request a snapshot. Hope this helps.

Hello @cedricve .
Thank you, very much for you support and quick replies.

I just spun up latest image of the Agent. I tried to lookup snapshot in the swagger API documentation.
If it is already implemented, could you please point me to a specific endpoint or write a quick and dirty url how to get a 
a snapshot? Otherwise, I'm looking forward to the update.

Kerberos.io is awesome system.

Thank you, very much.
Have a nice holidays.

cedricve, 2023-12-28:
@unichargepay feature added to separate branch, will be merged soon, and made available.

cedricve, 2024-01-03:
hello @unichargepay  made available in latest release: https://github.com/kerberos-io/agent/releases/tag/f739d52
Closing this request.

closed 2024-01-03 by cedricve