kerberos-io/agent #131: I need help with the documentation

Remmom, 2023-12-29:
Hi, I am new to Kerberos but have managed to install it and add some cameras. Your documentation seems to be limited to introducing the products you are offering and the installation process. I want to know if there is documentation on how to integrate Kerberos with 3rd party plugins and how to implement other services like motion detection?

cedricve, 2024-01-03:
hello @Remmom, thanks for your feedback! We definitely need to double down on documentation for custom development use cases. So far we have been actively developing on our own products, still keeping in mind to make it as open as possible. I believe some documentation and real examples (in the form of tutorials) would make a lot of sense and show-off the possibilities. I'll keep this issue open in the meantime.

Remmom, 2024-01-04:
Hi CedricThanks for the update.Right now I am testing Kerberos. I have installed Kerberos Factory and Kerberos Vault, but am having problems getting them to to communicate. I can add an Agent on Kerberos Factory but the Timezone facility in Global settings refuses to update to my Timezone. I am also not getting a feed from my camera, though it shows to be online.Kerberos Vault opens the login portal but becomes unresponsive after I enter the password and username. It then crashes and my kubernetes pod gives a 'Crashloopbackoff' error. I am working on Google Cloud Platform.Please advise. Best RegardsRemington

Remmom, 2024-01-04:
Hi Cedric

I see the Timezone function is now functioning properly. Thanks for sorting that out.

I have managed to connect to both Kerberos Factory and Kerberos Vault. I have set up the Cloud Storage and an account.

After successfullysetting up a camera in Kerberos Factory, I am getting this error message: '[31mERRO[0m[2024-01-04T17:42:57+02:00] cloud.HandleHeartBeat(): Disabled as we do not have a public key defined.
[31mERRO[0m[2024-01-04T17:43:07+02:00] cloud.HandleHeartBeat(): (400) Something went wrong while sending to Kerberos Vault.'

Please advice on next steps.

Kind Regards

Remington