kerberos-io/agent #132: Issue while verifying the camera settings: invalid Content-Base

CaptainCommuter, 2024-01-09:
In the latest version of the agent when verifying the connection, the error "Something went wrong while verifying the camera settings: invalid Content-Base" occurs.
This works OK with the November docker release and is running on Raspberry Pi and a S3VC IP camera.

cedricve, 2024-01-10:
Thanks for reporting @CaptainCommuter, we are using a new RTSP library. Which camera are you using?

cedricve, 2024-01-10:
can you share the RTSP url you are using? Are there other RTSP urls you can test out?

CaptainCommuter, 2024-01-10:
Thanks for responding. The camera is a generic one from SV3C (www.sv3c.com) A series and the RTSP urls are:
rtsp://ipcam01/stream0 for the main stream and
rtsp://ipcam01/stream1 for the sub-stream.
None work and unfortunately that's all there is.

cedricve, 2024-01-10:
The error thrown happens here: https://github.com/bluenviron/gortsplib/blob/main/client.go#L73-L88
I expect the RTSP url is not able to be parsed with our new library. Can you share the full log?
    
    if len(cb) != 1 {
	    return nil, fmt.Errorf("invalid Content-Base: '%v'", cb)
    }
    
    ret, err := base.ParseURL(cb[0])
    if err != nil {
	    return nil, fmt.Errorf("invalid Content-Base: '%v'", cb)
    }

I'm wondering if you use any special characters in the password? The RTSP url should work with following format (regex).
https://github.com/bluenviron/gortsplib/blob/main/pkg/base/url.go#L18-L45

cedricve, 2024-01-10:
Ok that's great news, but it should work without password as well (not advised though).

closed 2024-01-10 by cedricve