kerberos-io/agent #134: Audio issues with a reflashed Wyse camera

NichUK, 2024-01-24:
I'm just starting to test out Kerberos.io, and so far I like the simplicity of one agent per camera, and also the overall feel of the project (I'm looking at moving from Blue Iris, which I've used for the last five years).
However I'm having issues getting audio from my cameras (which worked in BI, and also works using VLC to look at the network stream directly). The codec options are OPUS, PCM, PCMU and MP3.
From other questions, it looks like PCM or PCMU should work, but neither of them seem to, either live, or in recordings, and OPUS just makes a horrible noise, and MP3 works, but I don't think you support it.
The codec information from VLC for the PCMU options is: 
[image]
Is this subtly different to a standard PCMU?
And for the PCM option is:
[image]

Do you have any ideas?

cedricve, 2024-01-24:
Hey there! Do you have G711 support, it would be ulaw I believe. To be complete, if you want audio in your recordings you'll need to select an AAC codec, if you want it to be in the livestream through Kerberos Hub, it has to be G711.

We are currently working on a transcoder and expanding audio codecs like OPUS. Hope this helps a bit..

cedricve, 2024-01-24:
Just looking at one of our cameras, I have following setup  with a sample rate of 8k instead of yours which is 16k.

[image]

NichUK, 2024-01-24:
That's a very similar looking codec to mine. 

Just to check, if I'm looking at the camera on the agent dashboard, should I expect audio there?

You also said that at present, only AAC is recorded, so we wouldn't expect the PCMU to make it to a recording until your transcoder is complete?

I'm just wondering suddenly if maybe it is working, but I'm just not looking at it somewhere that I should be able to hear it. Is there a streaming address on the agent that I can connect VLC to, or anything like that?