kerberos-io/agent #135: Can't use RTSP with UDP connection

mvdberge, 2024-02-14:
Error in Web GUI: "server wants to use the UDP transport protocol"

The RTSP client is fixed to TCP connections, see this line:
https://github.com/kerberos-io/agent/blob/3f29d1c46ff1f6557ac883f37b00e50c8bc2db9d/machinery/src/capture/gortsplib.go#L83

I think this line could just be deleted, as the documentation of gortsplib says that it has an automatism to choose the right transport variant (see https://github.com/bluenviron/gortsplib/blob/2c05e2d3ea16ae15d2bbc5846c6d823e3bcaed39/client.go#L228C1-L231C22):
```
        // transport protocol (UDP, Multicast or TCP).
        // If nil, it is chosen automatically (first UDP, then, if it fails, TCP).
        // It defaults to nil.
        Transport *Transport
```

cedricve, 2024-03-17:
I believe we might need an additional header to set/force TCP or UDP, but setting it to auto can give unwanted experiences. Let me know what you think @mvdberge

cedricve, 2024-07-04:
We'll add a method to choose for UDP, TCP or Auto