kerberos-io/agent #137: Recordings created but not display in web interface

thovu162020, 2024-02-19:
*Steps to reproduce*
- Setup a working camera with RTSP enabled.
- Follow the guide to setup [single docker container for Kerberos Agent](https://github.com/kerberos-io/agent/tree/master/deployments/docker#1-running-a-single-container). Image is `kerberos-agent:latest`
- Draw and save 'Region Of Interest' to enable motion detection. Verified the region of interest is saved to local `config.json`

*Expectations*
- At page 'Dashboard', values of 'Total recordings' and 'Latest events' should update accordingly, i.e. API `/api/dashboard` should have correct values of `latestEvents` and `numberOfRecordings`.
- At page 'Recordings', all recordings should be displayed, i.e. API `/api/latest-events` should have all event info

*Actual Observation*
- `Total recordings` has correct value, but `Latest events` are empty.
- Page `Recordings` are empty.

I checked and confirmed that the mp4 recordings are stored in directory `agent/recordings` and matches the value of  `numberOfRecordings` from API `/api/dashboard`

cedricve, 2024-02-19:
hello thank you for the details! Could you share your Kerberos Agents (docker logs) as well, so I can see a bit more context on your issue?

thovu162020, 2024-02-20:
@cedricve Sure, here are the logs from the agent. `LOG_LEVEL` is `"debug"`
```
INFO[2024-02-20T10:15:56+08:00] components.Kerberos.ControlAgent(): Number of packets read from mainstream: 170
INFO[2024-02-20T10:15:56+08:00] components.Kerberos.ControlAgent(): Number of packets read from substream: 170
[GIN] 2024/02/20 - 02:15:57 | 200 |     667.159µs |  129.126.84.234 | GET      "/api/dashboard"
ERRO[2024-02-20T10:15:59+08:00] cloud.HandleHeartBeat(): Disabled as we do not have a public key defined.
INFO[2024-02-20T10:15:59+08:00] cloud.HandleHeartBeat(): (200) Heartbeat received by Kerberos Vault.
[GIN] 2024/02/20 - 02:16:00 | 200 |     137.722µs |      10.0.17.31 | GET      "/"
INFO[2024-02-20T10:16:01+08:00] components.Kerberos.ControlAgent(): Number of packets read from mainstream: 173
INFO[2024-02-20T10:16:01+08:00] components.Kerberos.ControlAgent(): Number of packets read from substream: 172
INFO[2024-02-20T10:16:01+08:00] capture.main.HandleRecordStream(motiondetection): recording started
[GIN] 2024/02/20 - 02:16:02 | 200 |     673.498µs |  129.126.84.234 | GET      "/api/dashboard"
INFO[2024-02-20T10:16:03+08:00] capture.main.HandleRecordStream(motiondetection): motion detected while recording. Expanding recording.
INFO[2024-02-20T10:16:03+08:00] capture.main.HandleRecordStream(motiondetection): Received message with recording data, detected changes to save: 588
INFO[2024-02-20T10:16:05+08:00] capture.main.HandleRecordStream(motiondetection): motion detected while recording. Expanding recording.
INFO[2024-02-20T10:16:05+08:00] capture.main.HandleRecordStream(motiondetection): Received message with recording data, detected changes to save: 794
INFO[2024-02-20T10:16:06+08:00] components.Kerberos.ControlAgent(): Number of packets read from mainstream: 175
INFO[2024-02-20T10:16:06+08:00] components.Kerberos.ControlAgent(): Number of packets read from substream: 175
[GIN] 2024/02/20 - 02:16:07 | 200 |     650.186µs |  129.126.84.234 | GET      "/api/dashboard"
ERRO[2024-02-20T10:16:09+08:00] cloud.HandleHeartBeat(): Disabled as we do not have a public key defined.
INFO[2024-02-20T10:16:09+08:00] cloud.HandleHeartBeat(): (200) Heartbeat received by Kerberos Vault.
[GIN] 2024/02/20 - 02:16:10 | 200 |     148.063µs |      10.0.17.31 | GET      "/"
[GIN] 2024/02/20 - 02:16:10 | 200 |     120.452µs |       127.0.0.1 | GET      "/"
INFO[2024-02-20T10:16:11+08:00] components.Kerberos.ControlAgent(): Number of packets read from mainstream: 178
INFO[2024-02-20T10:16:11+08:00] components.Kerberos.ControlAgent(): Number of packets read from substream: 177
[GIN] 2024/02/20 - 02:16:12 | 200 |     671.086µs |  129.126.84.234 | GET      "/api/dashboard"
INFO[2024-02-20T10:16:16+08:00] components.Kerberos.ControlAgent(): Number of packets read from mainstream: 180
INFO[2024-02-20T10:16:16+08:00] components.Kerberos.ControlAgent(): Number of packets read from substream: 180
[GIN] 2024/02/20 - 02:16:17 | 200 |     541.966µs |  129.126.84.234 | GET      "/api/dashboard"
ERRO[2024-02-20T10:16:19+08:00] cloud.HandleHeartBeat(): Disabled as we do not have a public key defined.
INFO[2024-02-20T10:16:19+08:00] cloud.HandleHeartBeat(): (200) Heartbeat received by Kerberos Vault.
[GIN] 2024/02/20 - 02:16:20 | 200 |     150.043µs |      10.0.17.31 | GET      "/"
INFO[2024-02-20T10:16:21+08:00] components.Kerberos.ControlAgent(): Number of packets read from mainstream: 183
INFO[2024-02-20T10:16:21+08:00] components.Kerberos.ControlAgent(): Number of packets read from substream: 182
[GIN] 2024/02/20 - 02:16:22 | 200 |     656.375µs |  129.126.84.234 | GET      "/api/dashboard"
INFO[2024-02-20T10:16:23+08:00] capture.main.HandleRecordStream(motiondetection): closing recording (timestamp: 1708395365, recordingPeriod: 20, now: 1708395383, startRecording: 1708395352, maxRecordingPeriod: 30
INFO[2024-02-20T10:16:23+08:00] capture.main.HandleRecordStream(motiondetection): file save: 1708395352_6-967003_tsp_sg_cctv_01_200-200-400-400_2276_769.mp4
INFO[2024-02-20T10:16:26+08:00] components.Kerberos.ControlAgent(): Number of packets read from mainstream: 185
INFO[2024-02-20T10:16:26+08:00] components.Kerberos.ControlAgent(): Number of packets read from substream: 185
[GIN] 2024/02/20 - 02:16:27 | 200 |     654.366µs |  129.126.84.234 | GET      "/api/dashboard"
[GIN] 2024/02/20 - 02:16:29 | 200 |     658.875µs |  129.126.84.234 | GET      "/api/dashboard"
ERRO[2024-02-20T10:16:29+08:00] cloud.HandleHeartBeat(): Disabled as we do not have a public key defined.
INFO[2024-02-20T10:16:29+08:00] cloud.HandleHeartBeat(): (200) Heartbeat received by Kerberos Vault.
[GIN] 2024/02/20 - 02:16:30 | 200 |      149.52µs |      10.0.17.31 | GET      "/"
INFO[2024-02-20T10:16:31+08:00] components.Kerberos.ControlAgent(): Number of packets read from mainstream: 188
INFO[2024-02-20T10:16:31+08:00] components.Kerberos.ControlAgent(): Number of packets read from substream: 187
[GIN] 2024/02/20 - 02:16:32 | 200 |     675.016µs |  129.126.84.234 | GET      "/api/dashboard"
INFO[2024-02-20T10:16:36+08:00] components.Kerberos.ControlAgent(): Number of packets read from mainstream: 190
INFO[2024-02-20T10:16:36+08:00] components.Kerberos.ControlAgent(): Number of packets read from substream: 190
[GIN] 2024/02/20 - 02:16:37 | 200 |     664.203µs |  129.126.84.234 | GET      "/api/dashboard"
ERRO[2024-02-20T10:16:39+08:00] cloud.HandleHeartBeat(): Disabled as we do not have a public key defined.
INFO[2024-02-20T10:16:39+08:00] cloud.HandleHeartBeat(): (200) Heartbeat received by Kerberos Vault.
```

DedSecMF, 2025-03-04:
I am also facing the same error. Is there any fix for this?

DedSecMF, 2025-04-17:
@cedricve Can you take a look into this bug?

cedricve, 2025-04-17:
Hey @DedSecMF, thanks for the reminder here. Yes will have a look! keep you posted.

DedSecMF, 2025-06-20:
Hi @cedricve,
I debugged and found the issue.
The issue was with the name of my camera. If the camera name contains `_` in it, it would break.

[image]
We are splitting the file name according to `_` and my video recordings had an additional `_` due to my camera name.

So I just updated my camera name and now its working fine.

So we can add a note saying not to add `_` to AGENT_NAME.
AGENT_NAME	The agent friendly-name without `_`.	"agent"