kerberos-io/agent #141: Not sure how to connect camera

RickAndTired, 2024-06-07:
I'm not sure what I'm doing wrong, I have this rtsp url that I've tested and works in other NVR servers:
rtsp://192.168.1.160:1935/

When no username/password are set I get this error when using the rtsp url above:
`Something went wrong while verifying the camera settings: invalid Content-Base: '[192.168.1.160:1935/]'`

If I set a username and password on the rtsp stream - I don't see where to enter those in Kerberos and I get this error:
`Something went wrong while verifying the camera settings: bad status code: 401 (Unauthorized)`

Happens in both Kerberos docker and snap

RickAndTired, 2024-06-20:
> You could try `rtsp://user:password@192.168.1.xxx`

This one doesn't work either

cedricve, 2024-07-04:
@RickAndTired  what is the camera / model / brand you are using?

RickAndTired, 2024-07-07:
I was using an Android phone with the app RTSP Camera Server Pro.

I was only using it so I could test out Kerberos, but since it didn't work I ended up buying an IP camera.

closed 2024-07-07 by RickAndTired

cedricve, 2024-07-07:
Thanks for the update @RickAndTired!