kerberos-io/agent #145: External conditionnot working

enduluc, 2024-07-27:
For some reason, the external condition using a condition url stop working,.... I have several agents configured, when meets the Region of Interest, and the condition url gets the 200 Status code, I should start recording. I almost have 5 months with this configuration, but now stop working. even if I reinstall everything from scratch

cedricve, 2024-07-28:
hello @enduluc, I assume you are still using the old Kerberos Agent?

enduluc, 2024-07-29:
Version 3.1.8
I even reinstall everything, install new ones, and in one case, started working again, but when I restart the container, stopped ....The thing is that I have a nodered flow that enable an HTTP endpoint, when Im at home, the endpoint sends in the response (404) status code, when I leave, the endpoint sends a 200....the kerberos agent, verify the conditional URL, but even with the 200, its says is that the response is not 200, and stop the recording. 

I even tried entering to the containers´ console, test the command curl -i to the endpoint and the response is 200. but the agent says NOPE.

enduluc, 2024-08-02:
hi @cedricve any updates? do you need more information about this, logs, or anything else?  
I even install one agent in another environment, same results. The agent doesnt recognize the 200 status code from the endpoint response. 

Please let me know.

kind regards,

enduluc, 2024-08-03:
Hi @cedricve I have an update. I tested two agents with different url endpoints (condition url). Those endpoints are in my nodered instance. The results:

http://192.168.192.30:1880/endpoint    ----> this always works in keberos agent and the status code in browser is 200
http://192.168.192.30:1880/endpoint/other ----> this one doesn't work in kerberos agent, but still status code in browser is 200

http://192.168.192.30:1880/endpoint/  ----> when I put the "/" to the end, in kerberos agent stop working and says that response is not 200.

The condition URL in kerberos agent cannot validated the url response when you have another deep level in the url.
http://192.168.192.30:1880/endpoint     ----kerberos OK
http://192.168.192.30:1880/endpoint/   ----kerberos Failed
http://192.168.192.30:1880/endpoint/other   ----kerberos Failed
http://192.168.192.30:1880/endpoint/other/   ----kerberos Failed

cedricve, 2024-08-10:
Thanks for the information @enduluc, we will provide a fix! ;)

enduluc, 2024-09-15:
Hi @cedricve, any updates?

enduluc, 2024-10-06:
Hi @cedricve any updates? 

I ran some test and find out that kerberos only works with nodered home assistant addon. When i recreate the same flow in a nodered vm standalone, the endpoint returns 200 but kerberos says nop.

enduluc, 2024-10-17:
Hi @cedricve, is me again.
Just to know if there is light at the end of the tunnel. 
Im running a bunch of tests, with different node-red instances (lxc, docker, vm, home assistant OS) the only that works is with haos, with the main endpoint, but when I try to use a second endpoint, it fails. the workaround was installing two VMs with haos and run different endpoints.... please, need some light here!

enduluc, 2024-11-25:
Hi @cedricve I hope you are doing well. Have you had the oportunity to check this bug?

andp97, 2024-11-28:
Hi @enduluc,

I have looked at and tested the code responsible for this error message, but I am unable to reproduce this error:

> INFO[2024-07-27T14:20:06-04:00] conditions.uri.IsValidUriResponse(): response not 200, disabling recording.

Could you please share with us the output of these commands:
```
curl -I http://192.168.192.30:1880/endpoint
curl -I http://192.168.192.30:1880/endpoint/other
curl -I http://192.168.192.30:1880/endpoint/
curl -I http://192.168.192.30:1880/endpoint/other/
```
The Kerberos External Condition URI check will fail if these conditions are met:
- The remote endpoint returns an HTTP code strictly different from 200 (we can improve this by adding all 2xx status codes @cedricve)  
- The URL is invalid/malformed (you should see a frontend validation error)
- The remote host is unreachable

Based on the information that you provided, I suppose that your endpoint returns a different HTTP status code than 200 when a trailing slash is added to the URL. 

For example, your endpoint web server might be configured to redirect the URL to the non-trailing slashed version and return an HTTP 301 status code. In this case, the External Condition check will fail. 

I suggest you double-check the endpoint web server configuration and share the output of the above commands with us, so we can better handle this scenario.

enduluc, 2025-04-25:
ok, lets recapitulate:

agent and nodered installed in docker, same network

inside agent sh

**~ $ curl -i http://172.16.0.20:1880/endpoint
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
X-Content-Type-Options: nosniff
Content-Type: application/json; charset=utf-8
Content-Length: 2
ETag: W/"2-vyGp6PvFo4RvsFtPoIWeCReyIC8"
Date: Fri, 25 Apr 2025 00:58:24 GMT
Connection: keep-alive
Keep-Alive: timeout=5

~ $ curl -i http://172.16.0.20:1880/endpoint/
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
X-Content-Type-Options: nosniff
Content-Type: application/json; charset=utf-8
Content-Length: 2
ETag: W/"2-vyGp6PvFo4RvsFtPoIWeCReyIC8"
Date: Fri, 25 Apr 2025 00:58:27 GMT
Connection: keep-alive
Keep-Alive: timeout=5**

log from the agent:

**INFO[2025-04-24T21:02:57-04:00] conditions.uri.IsValidUriResponse(): response not 200, disabling recording.** 

url configured:

**ConditionURI: http://172.16.0.20:1880/endpoint**

enduluc, 2025-05-14:
any thoughts ??? @cedricve  @andp97

andp97, 2025-05-14:
Hi @enduluc,

As I mentioned earlier:

> The Kerberos External Condition URI check will fail if these conditions are met:
> - The remote endpoint returns an HTTP code strictly different from 200 [...]

Your endpoint is returning an HTTP 301 status code. For example:

```
~ $ curl -i http://10.0.0.16:1880/endpoint
HTTP/1.1 301 Moved Permanently
```

As shown in the `curl` response, you **must** point the external condition to the URL specified in the `Location` header (that URL will return an HTTP 200 status code).

Try using `http://10.0.0.16:1880/endpoint/` instead — note the trailing slash.

enduluc, 2025-05-15:
I fixed the 301. I test it within the sh container and the browser, got the 200.

But still, not working.

Easy way to replicate this:

Install kerberos agent on docker.
Install node-red (on docker, or linux, or pi, .. whatever you like)
Install in another vm/lxc, home assistant os and the node-red addon

In both node-red's, create a http endpoint. And make the response 200.

You will see it only works on node-red addon in home assistant, the others will fail.

cedricve, 2025-05-15:
Ok thanks, do you have some instructions so we can easily replicate. for example the docker run commands @enduluc ?

enduluc, 2025-05-15:
1. for the kerberos agents I use a compose and config files (check files attached):

[config.json](https://github.com/user-attachments/files/20229738/config.json)
[composefile.txt](https://github.com/user-attachments/files/20229739/composefile.txt)

2. For the homeassistant OS I use a vm. Install HOMEASSISTANT OS version, so you can use addons
https://www.home-assistant.io/installation/linux

3. Once inside Home assistant, install this addon: 
https://github.com/hassio-addons/addon-node-red/tree/v19.0.2

NOTE: in this particular addon, the httpNodeRoot is set fixed to `/endpoint` , so when you create it, the url is going to be http://youripaddress:1880/endpoint

4. Configure an HTTP endpoint, you can do it simple, and the default response is 200. If you want, check mine in the image attached. You will need some extra nodes to install within nodered
https://cookbook.nodered.org/http/create-an-http-endpoint

[image]

5. point the condition_url to the htttp endpoint created in nodered. 

In this escenario, it works!!!

BUT, when I install node-red outside Home Assistant and configuring everything right, it doesn{t work.

Results within the kerberos agent container:

THIS ONE IS FROM HOMEASSISTANT NODERED ADDON, IT WORKS, EVEN WITHOUT THE TRAILING SLASH AND THE 301
~ $ curl -I http://10.0.0.16:1880/endpoint/
HTTP/1.1 200 OK
Server: nginx
Date: Thu, 15 May 2025 14:43:59 GMT
Content-Type: application/json; charset=utf-8
Content-Length: 2
Connection: keep-alive
Access-Control-Allow-Origin: *
X-Content-Type-Options: nosniff
ETag: W/"2-vyGp6PvFo4RvsFtPoIWeCReyIC8"
Vary: Accept-Encoding
X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=block
X-Robots-Tag: none
Cache-Control: no-store, no-cache, must-revalidate, proxy-revalidate, max-age=0

THIS ONE IS FROM NODERED ON DOCKER, IT DOESN´T WORK, WITH OR WITHOUT TRAILING SLASH
~ $ curl -I http://10.50.0.11:1880/hello/
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Cache-control: no-store, no-cache, must-revalidate, proxy-revalidate, max-age=0
Content-Length: 85
Content-Type: text/html; charset=utf-8
ETag: W/"55-sNKc4nVHJOl00feMetgY4h5Gxv0"
Date: Thu, 15 May 2025 14:44:15 GMT
Connection: keep-alive
Keep-Alive: timeout=5

~ $ curl -I http://10.50.0.11:1880/hello
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Cache-control: no-store, no-cache, must-revalidate, proxy-revalidate, max-age=0
Content-Length: 85
Content-Type: text/html; charset=utf-8
ETag: W/"55-sNKc4nVHJOl00feMetgY4h5Gxv0"
Date: Thu, 15 May 2025 14:47:17 GMT
Connection: keep-alive
Keep-Alive: timeout=5

LOGS FROM THE KERBEROS AGENT:
WITH THE 10.0.0.16. HOME ASSISTANT AND NODERED ADDON:
INFO[2025-05-15T10:48:01-04:00] conditions.uri.IsValidUriResponse(): response 200, enabling recording. 

WITH THE 10.50.0.11. NODERED on docker:
INFO[2025-05-15T10:50:52-04:00] conditions.uri.IsValidUriResponse(): response not 200, disabling recording.

enduluc, 2025-05-17:
@cedricve have you ever consider to use multiple methods for the external condition. Like, using mqtt or a native logic funtion that can be program or configured, or by integrations with programs like home assistant, node-red or n8n. It would be awesome.

This problem, i think is on the red-node side, and probably the addon version has something that simply works. its weird that in a clean installation of node-red and kerberos agent, dont work even knowing that the endpoint is correctly configured. With the same steps that I did on the addon instance.

In the mean time, I did a workaround, having two instances of home assistants with node-red addons, so I can use two flows (actual requirement to my setup), so one group of agents use one flow and another group of agents uses the other flows. The bad thing is that I need to waste more resources in order to mantain both instances running. 

Please, thinks about the idea. multiple methods, native flows function, or full integration with popular platforms like home assistant, node-red or n8n.  Still, your project rocks just like it is, so congrats. 

Thanks in advanced. 👍