kerberos-io/agent #147: run from source code

rotragit, 2024-08-20:
I'm trying to run locally from source code, but I get:

src/onvif/main.go:1264:38: relayoutputs.RelayOutputs.Token undefined (type []"github.com/kerberos-io/onvif/xsd/onvif".RelayOutput has no field or method Token)

rotragit, 2024-08-20:
`git clone https://github.com/kerberos-io/agent.git
cd agent/machinery
./update-mod.sh
go run main.go -action run -port 1234`

output is:

`$ go run main.go -action run -port 1234
# github.com/kerberos-io/agent/machinery/src/onvif
src/onvif/main.go:1264:38: relayoutputs.RelayOutputs.Token undefined (type []"github.com/kerberos-io/onvif/xsd/onvif".RelayOutput has no field or method Token)
`

rotragit, 2024-08-21:
Same running build.sh, btw:

``
=> CACHED [build-ui 4/7] COPY ui /go/src/github.com/kerberos-io/agent/ui                                                                                                                                                               0.0s
 => ERROR [build-machinery  9/13] RUN cd /go/src/github.com/kerberos-io/agent/machinery &&  go mod download &&  go build -tags timetzdata,netgo,osusergo --ldflags '-s -w -extldflags "-static -latomic"' main.go &&  mkdir -p /agent  74.0s
 => CANCELED [build-ui 5/7] RUN cd /go/src/github.com/kerberos-io/agent/ui && rm -rf yarn.lock && yarn config set network-timeout 300000 &&  yarn && yarn build                                                                        76.4s
------                                                                                                                                                                                                                                       
 > [build-machinery  9/13] RUN cd /go/src/github.com/kerberos-io/agent/machinery &&     go mod download &&      go build -tags timetzdata,netgo,osusergo --ldflags '-s -w -extldflags "-static -latomic"' main.go &&    mkdir -p /agent &&  mv main /agent &&        mv version /agent &&    mv data /agent &&       mkdir -p /agent/data/cloud &&   mkdir -p /agent/data/snapshots &&       mkdir -p /agent/data/log &&     mkdir -p /agent/data/recordings &&      mkdir -p /agent/data/capture-test &&         mkdir -p /agent/data/config &&  rm -rf /go/src/gitlab.com/:                                                                                                                                                          
60.28 # github.com/kerberos-io/agent/machinery/src/onvif                                                                                                                                                                                     
60.28 src/onvif/main.go:1264:38: relayoutputs.RelayOutputs.Token undefined (type []"github.com/kerberos-io/onvif/xsd/onvif".RelayOutput has no field or method Token)                                                                        
------                                     
``

cedricve, 2024-08-27:
hey @rotragit, make sure the latest onvif library is used.
https://github.com/kerberos-io/agent/blob/master/machinery/go.mod#L24