kerberos-io/agent #148: Docker run behind proxy

mattgaviota, 2024-10-10:
There is a way to setup kerberos agent to work behind a nginx proxy? i tried differents configuration on nginx without luck.

cedricve, 2024-10-10:
hey @mattgaviota  can you elaborate a bit more on the use case?

mattgaviota, 2024-10-15:
Yes @cedricve, thanks for asking.

I have been trying kerberos agents on docker with kerberos hub pro. For that, i used a docker compose with the agents i needed, then with a nginx i made a reverse proxy to had agent1.mydomain.com, agent2.mydomain.com and so on.

This solution works perfectly, but i noticed that if i want to add more agents, i'll have to do all the configuration, including a new service, new server and a new subdomain. 

For those reason i tried to make a single server on nginx, and had different location for different agents, something like agents.mydomain.com/agent1, agents.mydomain.com/agent2 and so on, and at least avoid the subdomain part. And that configuration i couldn't make it work because i  always miss something, 404 errors, redirects, etc.

Thanks in advance, sorry if my english is not clear

andp97, 2024-11-28:
Hi @mattgaviota,  

I encountered a similar issue with 404 errors and redirects when setting up a reverse proxy.  

Have you tried setting the `Host` header to ensure proper routing?  

Here’s an example NGINX configuration (converted from my setup using Caddy as a reverse proxy):  

```nginx
server {
    listen 80;
    server_name agents.mydomain.com;

    location /agent1/ {
        proxy_pass http://agent1.internal/;
        proxy_set_header Host $host;  # Prevent redirect issues
        
        # Additional configuration...
    }

    location /agent2/ {
        proxy_pass http://agent2.internal/;
        proxy_set_header Host $host;  # Prevent redirect issues
        
        # Additional configuration...
    }
}
```  

Let me know if this works for your setup!