kerberos-io/agent #160: Feature Request: Docker Compose - .env example for testing purposes

NathanGilbert2, 2025-01-27:
As an alt to pushing the config file, it would be nice to have a .env file in your example for launching via [docker compose](https://github.com/kerberos-io/agent/tree/master/deployments/docker#2-running-multiple-containers-with-docker-compose) with all the available environment variables prepopulated for quick modification during initial testing. It is also good for sharing configurations with multiple agents as I was using it for.

PS.
Just had to add `env_file: ".env"` to each agent service definition in my docker compose file and this worked great for me. (Normally I don't believe this is required as docker compose just looks for .env in the project directory, so I will say this is the fun of testing on Windows.)

```yaml
services:
  kerberos-agent1:
    image: "kerberos/agent:latest"
    ports:
      - "8081:80"
    environment:
      AGENT_NAME: agent1
      AGENT_KEY: "1"
      AGENT_CAPTURE_IPCAMERA_ONVIF_XADDR: x.x.x.x:8899
      AGENT_CAPTURE_IPCAMERA_RTSP: rtsp://x:x@x.x.x.x:554/live/ch0
      AGENT_CAPTURE_IPCAMERA_SUB_RTSP: rtsp://x:x@x.x.x.x:554/live/ch1
    volumes:
      - ./agent1/recordings:/home/agent/data/recordings
    env_file: ".env"
  kerberos-agent2:
    image: "kerberos/agent:latest"
    ports:
      - "8082:80"
    environment:
      AGENT_NAME: agent2
      AGENT_KEY: "2"
      AGENT_CAPTURE_IPCAMERA_ONVIF_XADDR: x.x.x.x:8899
      AGENT_CAPTURE_IPCAMERA_RTSP: rtsp://x:x@x.x.x.x:554/live/ch0
      AGENT_CAPTURE_IPCAMERA_SUB_RTSP: rtsp://x:x@x.x.x.x:554/live/ch1
    volumes:
      - ./agent2/recordings:/home/agent/data/recordings
    env_file: ".env"
```

```yaml
# Level for logging, could be "info", "warning", "debug", "error" or "fatal". (Default: "info")
LOG_LEVEL=

# Logging output format "json" or "text". (Default: "text")
LOG_OUTPUT=

# You can choose to run this in 'release' for production, and or 'demo' for showcasing. (Default: "release")
AGENT_MODE=

# Specify if you want to use InsecureSkipVerify for the internal HTTP client. (Default: "false")
AGENT_TLS_INSECURE=

# The username used to authenticate against the Kerberos Agent login page. (Default: "root")
AGENT_USERNAME=

# The password used to authenticate against the Kerberos Agent login page. (Default: "root")
AGENT_PASSWORD=

# A unique identifier for your Kerberos Agent, this is auto-generated but can be overriden. (Default: "")
AGENT_KEY=

# The agent friendly-name. (Default: "agent")
AGENT_NAME=

# Timezone which is used for converting time. (Default: "Africa/Ceuta")
AGENT_TIMEZONE=

# When enabled, recordings uploaded successfully to a storage will be removed from disk. (Default: "true")
AGENT_REMOVE_AFTER_UPLOAD=

# Makes sure no external connection is made. (Default: "false")
AGENT_OFFLINE=

# Cleans up the recordings directory. (Default: "true")
AGENT_AUTO_CLEAN=

# If AUTO_CLEAN enabled, set the max size of the recordings directory (in MB). (Default: "100")
AGENT_AUTO_CLEAN_MAX_SIZE=

# Enable the timetable for Kerberos Agent (Default: "false")
AGENT_TIME=

# A (weekly) time table to specify when to make recordings "start1,end1,start2,end2;start1.. (Default: "")
AGENT_TIMETABLE=

# A single polygon set for motion detection: "x1,y1;x2,y2;x3,y3;... (Default: "")
AGENT_REGION_POLYGON=

# Full-HD RTSP endpoint to the camera you're targeting. (Default: "")
AGENT_CAPTURE_IPCAMERA_RTSP=

# Sub-stream RTSP endpoint used for livestreaming (WebRTC). (Default: "")
AGENT_CAPTURE_IPCAMERA_SUB_RTSP=

# Mark as a compliant ONVIF device. (Default: "")
AGENT_CAPTURE_IPCAMERA_ONVIF=

# ONVIF endpoint/address running on the camera. (Default: "")
AGENT_CAPTURE_IPCAMERA_ONVIF_XADDR=

# ONVIF username to authenticate against. (Default: "")
AGENT_CAPTURE_IPCAMERA_ONVIF_USERNAME=

# ONVIF password to authenticate against. (Default: "")
AGENT_CAPTURE_IPCAMERA_ONVIF_PASSWORD=

# Toggle for enabling or disabling motion. (Default: "true")
AGENT_CAPTURE_MOTION=

# Toggle for enabling or disabling liveview. (Default: "true")
AGENT_CAPTURE_LIVEVIEW=

# Toggle for enabling or disabling snapshot generation. (Default: "true")
AGENT_CAPTURE_SNAPSHOTS=

# Toggle for enabling making recordings. (Default: "true")
AGENT_CAPTURE_RECORDING=

# Toggle for enabling continuous "true" or motion "false". (Default: "false")
AGENT_CAPTURE_CONTINUOUS=

# If CONTINUOUS set to false, specify the recording time (seconds) before/after motion event. (Default: "10")
AGENT_CAPTURE_PRERECORDING=

# If CONTINUOUS set to false, specify the recording time (seconds) after motion event. (Default: "20")
AGENT_CAPTURE_POSTRECORDING=

# The maximum length of a single recording (seconds). (Default: "30")
AGENT_CAPTURE_MAXLENGTH=

# If CONTINUOUS set to false, the number of pixel require to change before motion triggers. (Default: "150")
AGENT_CAPTURE_PIXEL_CHANGE=

# Set the format of the recorded MP4 to fragmented (suitable for HLS). (Default: "false")
AGENT_CAPTURE_FRAGMENTED=

# If AGENT_CAPTURE_FRAGMENTED set to true, define the duration (seconds) of a fragment. (Default: "8")
AGENT_CAPTURE_FRAGMENTED_DURATION=

# An MQTT broker endpoint that is used for bi-directional communication (live view, onvif, etc) (Default: "tcp://mqtt.kerberos.io:1883")
AGENT_MQTT_URI=

# Username of the MQTT broker. (Default: "")
AGENT_MQTT_USERNAME=

# Password of the MQTT broker. (Default: "")
AGENT_MQTT_PASSWORD=

# If AGENT_REALTIME_PROCESSING set to true, the agent will send key frames to the topic (Default: "")
AGENT_REALTIME_PROCESSING=

# The topic to which keyframes will be sent in base64 encoded format. (Default: "")
AGENT_REALTIME_PROCESSING_TOPIC=

# When using WebRTC, you'll need to provide a STUN server. (Default: "stun:turn.kerberos.io:8443")
AGENT_STUN_URI=

# Force using a TURN server, by generating relay candidates only. (Default: "false")
AGENT_FORCE_TURN=

# When using WebRTC, you'll need to provide a TURN server. (Default: "turn:turn.kerberos.io:8443")
AGENT_TURN_URI=

# TURN username used for WebRTC. (Default: "username1")
AGENT_TURN_USERNAME=

# TURN password used for WebRTC. (Default: "password1")
AGENT_TURN_PASSWORD=

# Store recordings in Kerberos Hub (s3), Kerberos Vault (kstorage), or Dropbox (dropbox). (Default: "s3")
AGENT_CLOUD=

# Turning on/off encryption of traffic from your Kerberos Agent to Kerberos Hub. (Default: "true")
AGENT_HUB_ENCRYPTION=

# The Kerberos Hub API, defaults to our Kerberos Hub SAAS. (Default: "https://api.hub.domain.com")
AGENT_HUB_URI=

# The access key linked to your account in Kerberos Hub. (Default: "")
AGENT_HUB_KEY=

# The secret access key linked to your account in Kerberos Hub. (Default: "")
AGENT_HUB_PRIVATE_KEY=

# The Kerberos Hub region, to which you want to upload. (Default: "")
AGENT_HUB_REGION=

# The site ID of a site you've created in your Kerberos Hub account. (Default: "")
AGENT_HUB_SITE=

# The Kerberos Vault API url. (Default: "https://vault.domain.com/api")
AGENT_KERBEROSVAULT_URI=

# The access key of a Kerberos Vault account. (Default: "")
AGENT_KERBEROSVAULT_ACCESS_KEY=

# The secret key of a Kerberos Vault account. (Default: "")
AGENT_KERBEROSVAULT_SECRET_KEY=

# A Kerberos Vault provider you have created (optional). (Default: "")
AGENT_KERBEROSVAULT_PROVIDER=

# The directory, in the Kerberos vault, where the recordings will be stored. (Default: "")
AGENT_KERBEROSVAULT_DIRECTORY=

# The Access Token from your Dropbox app, that is used to leverage the Dropbox SDK. (Default: "")
AGENT_DROPBOX_ACCESS_TOKEN=

# The directory, in Dropbox, where the recordings will be stored. (Default: "")
AGENT_DROPBOX_DIRECTORY=

# Enable 'true' or disable 'false' end-to-end encryption for MQTT messages. (Default: "false")
AGENT_ENCRYPTION=

# Enable 'true' or disable 'false' end-to-end encryption for recordings. (Default: "false")
AGENT_ENCRYPTION_RECORDINGS=

# The fingerprint of the keypair (public/private keys), so you know which one to use. (Default: "")
AGENT_ENCRYPTION_FINGERPRINT=

# The private key (asymmetric/RSA) to decrypt and sign requests sent over MQTT. (Default: "")
AGENT_ENCRYPTION_PRIVATE_KEY=

# The symmetric key (AES) to encrypt and decrypt requests sent over MQTT. (Default: "")
AGENT_ENCRYPTION_SYMMETRIC_KEY=
```

cedricve, 2025-01-28:
Thanks for reporting @NathanGilbert2! @KilianBoute what do you think of this approach I know we used the template approach.

https://github.com/kerberos-io/agent/blob/master/deployments/docker/docker-compose.yaml#L2-L6

As indicated by @NathanGilbert2 I do think the `env_file` is more clean.

KilianBoute, 2025-01-28:
Yeah, might as well provide the option for both.