kerberos-io/agent #170: cloud.VerifyPersistence(kerberosvault): Something went wrong while verifying the persistence from the Agent

asinorum, 2025-03-28:
When attempting to connect Kerberos Agent v3.1 to Kerberos Vault v1.0 (both running in Docker), the persistence verification from the Agent fails with the error:

> Something went wrong while verifying the persistence :cloud.VerifyPersistence(kerberosvault): Something went wrong while verifying your persistence settings...

Agent Configuration uses the same settings as the Hub (which works):
Vault API URL: e.g., http://vault.mydomain.ltd/api.
Provider Name: Matches the storage provider configured in the Vault.
Username: Same as the Hub’s username.
Keys: Valid Access Key and Secret Key (identical to the Hub).

The Hub connects successfully to the Vault.
The Agent fails verification but creates empty files (0GB) in the Vault.
No network errors or explicit 404s were mentioned, only the generic error above.

Could you help identify potential misconfigurations or bugs causing this? 

Thank you!

cedricve, 2025-04-23:
Hello @asinorum, we are looking into this and will come back to you shortly.

asinorum, 2025-04-24:
Hi @cedricve 
Yes, the agent is recording and storing correctly. 
You can find all my tests and logs:
https://github.com/kerberos-io/vault/issues/23#issue-2954769367

As far as I could tell, there is an issue with Storj Storage, It works with GCP. I've opened a discussion thread in Storj's forum.
https://forum.storj.io/t/x-amz-content-sha256-header-mismatch-when-uploading-objects-to-kerberos-vault/29667

cedricve, 2025-04-24:
Thanks for the additional details @asinorum, we will verify. Keep you posted!

closed 2025-05-20 by cedricve