kerberos-io/agent #173: ONVIF motion events to trigger recording

OttoHollmann, 2025-04-14:
Hi there,

I’d like to know if kerberos-agent can be configured to use ONVIF motion events to trigger recordings.

I've successfully set up ONVIF in Kerberos, but recordings are not being triggered.

To verify, I used a simple script with the node-onvif-events library and confirmed that my camera is indeed sending ONVIF motion events.

I'm using the latest Docker ARM image, and I've configured the following environment variables:
- AGENT_NAME
- AGENT_CAPTURE_IPCAMERA_RTSP
- AGENT_CAPTURE_IPCAMERA_ONVIF_XADDR
- AGENT_CAPTURE_IPCAMERA_ONVIF_USERNAME
- AGENT_CAPTURE_IPCAMERA_ONVIF_PASSWORD

Am I missing something in the configuration, or is this feature currently not implemented?

Thanks in advance!

OttoHollmann, 2025-04-23:
Thank you for the confirmation.

Yes, you’ve understood it correctly. I’d like to use the camera’s motion detection (since it offers advanced motion settings with hardware acceleration) and run Agent on the Raspberry Pi as a lightweight recorder.

Would it be straightforward to implement? If so, I’ll give it a try.

cedricve, 2025-04-23:
cool, we have an onvif library here: https://github.com/kerberos-io/onvif. There is an event handler which we are currently using to read IO inputs and output values. Which is happening here: https://github.com/kerberos-io/agent/blob/master/machinery/src/cloud/Cloud.go#L299-L404

I would expect that ONVIF will push a message on the queue if motion is detected, if we could retrieve that message we should start a recording by push an event in motion channel: https://github.com/kerberos-io/agent/blob/master/machinery/src/computervision/main.go#L164-L168

This will then trigger a recording. I think all is possible, it is just a matter to know how to retrieve the ONVIF event (pullmessagepoint is just my assumption)

cedricve, 2025-05-22:
hey @OttoHollmann was wondering if you did some magic here? Happy to assist.